FIRST-Tech-Challenge/FtcRobotController #218: Missing Package in OnBoard Java Robot Controller 7.0

huckshold, 2021-10-30:
I am generating the following every time I try a Build with Version 7.0:

org/firstinspires/ftc/teamcode/UsRange.java line 46, column 55: ERROR: package
org.firstinspires.ftc.robotcore.external.analog does not exist

This message was generated when performing a Build on the ImuExercise.java Sample Program without modifications.

Any suggestions would be appreciated since the Team does not want to go back to using Blocks.

cmacfarl, 2021-10-30:
Can you post both the full UsRange.java and ImuExercise.java files online someplace?  Either in a github fork, or a gist, or some other easily readable format?

huckshold, 2021-10-30:
Problem resolved.  I noticed that "USRange.java" was a routine I created with version 6.0 of the Robot Controller.  It contained the reference the to "external.analog" package.  I removed that reference from the routine that I was not trying to build, and the error message disappeared.  I would infer that the "external.analog" package was judged to be unnecessary in version 7.0.  It is strange that the build process is referencing other routines unless I accidentally reused the name USRange.java.

NoahAndrews, 2021-11-11:
> It is strange that the build process is referencing other routines unless I accidentally reused the name USRange.java.

OnBotJava builds all files together, not just the one that's currently open.