FIRST-Tech-Challenge/FtcRobotController #221: TFOD Load Model from Asset

mihir-chauhan, 2021-10-31:
Hello,

We have a custom trained Tensorflow (v2.6) model that we want to use in the app to do Object Detection and ran into the following error when invoking: 

```
tfod.loadModelFromAsset(TFOD_MODEL_ASSET, LABEL_FIRST_ELEMENT);
```

When calling this function during the initialization part, it throws the following error:

```
E/AndroidRuntime: FATAL EXCEPTION: LinearOpMode main
    Process: com.qualcomm.ftcrobotcontroller, PID: 5528
    java.lang.NoSuchMethodError: No direct method <init>(Ljava/nio/ByteBuffer;I)V in class Lorg/tensorflow/lite/Interpreter; or its super classes (declaration of 'org.tensorflow.lite.Interpreter' appears in /data/app/com.qualcomm.ftcrobotcontroller-2/base.apk:classes3.dex)
        at org.firstinspires.ftc.robotcore.internal.tfod.TfodFrameManager1.<init>(TfodFrameManager1.java:46)
        at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.newTfodFrameManager(TFObjectDetectorImpl.java:305)
        at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.initialize(TFObjectDetectorImpl.java:284)
        at org.firstinspires.ftc.robotcore.internal.tfod.TFObjectDetectorImpl.loadModelFromAsset(TFObjectDetectorImpl.java:251)
        at org.firstinspires.ftc.teamcode.Autonomous.TensorFlow_Testing.initTfod(TensorFlow_Testing.java:132)
        at org.firstinspires.ftc.teamcode.Autonomous.TensorFlow_Testing.runOpMode(TensorFlow_Testing.java:53)
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper$1.run(LinearOpMode.java:279)
        at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
        at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode$LinearOpModeHelper.run(LinearOpMode.java:273)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
        at java.lang.Thread.run(Thread.java:761)
```

Please let me know if I am doing something wrong or if this may be a bug. Thanks!

Windwoes, 2021-10-31:
Does it still crash when using the stock model?

mihir-chauhan, 2021-11-30:
@Windwoes Thanks for the quick response, I think the problem was the model I was using, and this problem is resolved. Thanks again for your help!

closed 2021-11-30 by mihir-chauhan