FIRST-Tech-Challenge/FtcRobotController #232: Camera Stream option not available on Driver Station menu

shelbyRobots, 2021-11-12:
We use Vuforia and then grab the images and pass them to our openCV processing.
Last year, if we were using an opMode that initialized vuforia, we were able to preview the image from the webcam attached to the control hub via a menu option on the Driver's station.
This year, using identical vuforia init, the option isn't showing up.
I know that the vufroria init, and pass thru to openCV is working, because our processing pipeline runs and I save off the image.
Please help - we have a competition tomorrow.

Windwoes, 2021-11-13:
Are you running the OpMode? It's only available during init, not after you press play.

shelbyRobots, 2021-11-13:
I have only tried in init.  I know that it is not available either before init or after play.

WestsideRobotics, 2021-11-13:
A simplistic suggestion: start by testing the Sample OpMode named ConceptTensorFlowObjectDetectionWebcam.  Verify that Camera Stream appears in the DS main menu, and that the preview works for you.

Then start converting that Sample into your OpMode, section by section.  At some point you may discover which section of your code is causing Camera Stream to drop off the DS menu.

This assumes all devices are up to date with all software, verified by the REV Hardware Client.  Use the Sample OpMode from the 7.0 SDK.

shelbyRobots, 2021-11-13:
Everything is up to date.  I think Driver Hubs (new this year) updated to 7.0.1 by RevHardwareClient.  And the loading of vuforia is done by a class that hasn't change since it was working last year.
But great thought on using a sample opmode - not sure why I didn't think to try that.

shelbyRobots, 2021-11-14:
Not sure what we were doing wrong,  but camera stream was there today. All is good

closed 2021-11-14 by shelbyRobots