FIRST-Tech-Challenge/FtcRobotController #233: PwmControl.isPwmEnabled crashes

shelbyRobots, 2021-11-12:
I am able to use a  CRServoImplEx instance to use setPwmEnable/Disable.
However, isPwmEnabled crashes the app with the following:
`E EventLoopManager: com.qualcomm.robotcore.exception.RobotCoreException: EventLoop Exception in loop(): IllegalArgumentException - invalid pwm channel: 4`

Steps:
crsix.setPwmDisable();
...do other stuff...
if(crsix.setPwmDisable) crsix.setPwmDisable();

I was able to do a work around by keeping a local variable of pwm energize state when I call disable - to avoid calling enable on an already pwm energized port.

ServoControllerEx with port number parameter seemed to give same results.

Windwoes, 2021-11-13:
Looks to be a bug in the `LynxGetServoEnableCommand` constructor...

It reads

```java
    public LynxGetServoEnableCommand(LynxModuleIntf module, int channelZ)
        {
        this(module);
        LynxConstants.validatePwmChannelZ(channelZ);
        this.channel = (byte)channelZ;
        }
```

But it should be 
```java
    public LynxGetServoEnableCommand(LynxModuleIntf module, int channelZ)
        {
        this(module);
        LynxConstants.validateServoChannelZ(channelZ);
        this.channel = (byte)channelZ;
        }
```
```

Windwoes, 2021-12-21:
This will be fixed in the next release of the SDK.

Windwoes, 2022-01-17:
Fixed in 7.1

closed 2022-01-17 by Windwoes