FIRST-Tech-Challenge/FtcRobotController #238: TensorFlowObjectDetectionFreightFrenzy not found

baqwas, 2021-11-19:
Hello!

I can confirm that REV Hardware Client reports the Control Hub application & firmware are the latest release versions. Unfortunately, in Blocks IDE, I cannot select **TensorFlowObjectDetectionFreightFrenzy** after starting from **ConceptTensorFlowObjectDetectionWebcam** as the template.

Must be some operator/user _newbie_ error on my part. How can I correct the mistake, please? Thanks.

Kind regards.

WestsideRobotics, 2021-11-19:
When you **Create New OpMode**, using **ConceptTensorFlowObjectDetectionWebcam** as the Sample, there is nothing left to select -- that OpMode is ready to Save, then INIT and Start on the DS device.  It just runs, providing telemetry info from the image seen by the configured **Webcam 1**.  Assuming your RC app is version 7.0, that OpMode can recognize various game elements of Freight Frenzy.

From that point, what is your programming goal?

baqwas, 2021-11-19:
Thx @WestsideRobotics for the quick response.

The initial goal, of course, just determine the barcode _number_ for use as the Hub Level _number_. Then the rest of the goals will unfold in due course. Thanks.

Kind regards.

closed 2021-11-19 by baqwas

WestsideRobotics, 2021-11-19:
The FTC wiki has an explanation of that sample OpMode, with some hints for expanding on it for Freight Frenzy autonomous:

https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Blocks-Sample-Op-Mode-for-TensorFlow-Object-Detection

baqwas, 2021-11-19:
Thx again, @WestsideRobotics! It was the wiki that "drove" me to look at the samples and to post an inquiry as an issue. Feel bad about cluttering issues in GitHub in this manner but I wasn't getting any response from the various forums.

My handicap is two-fold: I'm mentoring the team remotely (with zero chance of getting my hands on a Control Hub before Dec per REV Robotics) and my reflex attitude is to confine myself simply to TF. Goes with my rookie mentor status with a rookie team! :)

You've been a big help in reducing my comprehension challenges. I'll visit the team over the weekend and explain my findings to the team. They were lucky to participate in the ML _beta_ and are now very enthusiastic about the potential.

Kind regards.

WestsideRobotics, 2021-11-19:
If you are mentoring remotely and without a Control Hub, I think it's fine to learn TFOD with only a pair of DS/RC phones for now (assuming you have those).  Use the RC phone's camera, and the sample called ConceptTensorFlowObjectDetection.  It all works the same as the webcam version.  Better for learning, in fact, since the RC preview is live video, and large.

Right now many teams are addressing how to adapt the sample OpMode for Freight Frenzy autonomous.  The sample is purposely intended to demonstrate TFOD without supplying a match-ready solution.

For adapting the sample, the key is to understand how it works (including its for-loop), then decide how you wish to exit its main forever loop.  Namely, what are your own criteria for "I know where the Duck is"?  I just met with a team today on this topic.

Here's a team working through this question, using the exact same Java version of the Blocks sample.  Scroll down for my comments to them.
https://www.reddit.com/r/FTC/comments/qurrtj/where_to_put_code_after_detection_with_tfod/

Good luck!

baqwas, 2021-11-19:
With respect to "_I know where the Duck is_" my _unproven_ initial thought is to use the detection box of the classified (_viz._ Duck) object that could serve the loop exit criterion. Some of the inline comments in the starter template code about depth of field and aspect ratio have been noted.

A side observation that I want to share with you is that there are only two dedicated programmers in the team. Yet six other team members partially stepped away from their other roles to evaluate the FTC _beta_. I hope that these TFOD exercises will provide another way to get them hooked for longer assignments.

baqwas, 2021-11-19:
Hello @WestsideRobotics,

Need one more piece of advice, please. I have to stay with Blocks for the team this season for many practical purposes. Once the competitions end, we will migrate to one of the other two options.

Right now I am basically _reverse engineering_ the code into Blocks (reading Java and ensuring that the flow and processes are similar to the extent practical). In my simplistic world, I cannot see any roadblocks. Given your expertise, do you see any pitfalls for our tight schedule? 

The team hasn't thought much about ML beyond the barcode exercises. For example, being able to leverage navigation permits the operators to stay focused narrowly and to gain a few more precious seconds. We weren't very successful with Light, Medium and Heavy Freight object classification in the _beta_. Don't quite have a feel on how to leverage this aspect without operator interactivity. Thanks again.

Kind regards.

WestsideRobotics, 2021-11-19:
Your careful approach should work fine for your team.  Your 'reverse engineering' effort will be helped by also using Blocks' built-in Java preview window.

I believe the vast majority of FTC tasks that can be done in Java, can be done in Blocks.  Gaps can be filled with [custom myBlocks](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Custom-FTC-Blocks-(myBlocks)) and [external libraries](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/External-Libraries-in-OnBot-Java-and-Blocks).

Would love to see teams benefiting from the field navigation targets.  Good learning opportunity there.

I have wondered about teams hoping to use vision to distinguish the weighted cubes.  Can automation yield better results than a human operator under high workload and obstructed line-of-sight?  That's quite a challenge; can't wait to see the solutions!