FIRST-Tech-Challenge/FtcRobotController #239: RC App Crashes on Launch and then Relaunches, not allowing debugging through ADB

mihir-chauhan, 2021-11-22:
Hello,

There seems to be a problem where when I use ADB through WiFi to install onto the Control Hub, the app crashes, and then it relaunches, allowing the DC to still connect, but makes it impossible to debug through my laptop since the app process ended and I no longer get logs back from the newly launched app. This problem is repeatable after restarting the control hub several times. Please let me know how to solve this. Thanks!

Here is the whole log causing the initial crash:

```
I/art: Rejecting re-init on previously-failed class java.lang.Class<com.qualcomm.robotcore.wifi.WifiDirectAssistant$1>: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/net/wifi/p2p/WifiP2pManager$DeviceInfoListener;
        at java.lang.Class java.lang.VMClassLoader.findLoadedClass!(java.lang.ClassLoader, java.lang.String) (VMClassLoader.java:-2)
        at java.lang.Class java.lang.ClassLoader.findLoadedClass(java.lang.String) (ClassLoader.java:742)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:362)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:367)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
I/art:     at java.lang.Class java.lang.Class.classForName!(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:-2)
        at java.lang.Class java.lang.Class.forName(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:400)
        at java.util.List org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(java.util.Collection) (ClassManager.java:214)
        at void org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses() (ClassManager.java:257)
        at void com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses() (ClassManagerFactory.java:94)
        at void org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(android.os.Bundle) (FtcRobotControllerActivity.java:356)
        at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:6709)
        at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1118)
        at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:2619)
        at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:2727)
        at void android.app.ActivityThread.-wrap12(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
        at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1478)
        at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:102)
        at void android.os.Looper.loop() (Looper.java:154)
        at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6121)
        at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
        at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:905)
        at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:795)
    Caused by: java.lang.ClassNotFoundException: Didn't find class "android.net.wifi.p2p.WifiP2pManager$DeviceInfoListener" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm64, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]
        at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:56)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:380)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
        at java.lang.Class java.lang.VMClassLoader.findLoadedClass!(java.lang.ClassLoader, java.lang.String) (VMClassLoader.java:-2)
        at java.lang.Class java.lang.ClassLoader.findLoadedClass(java.lang.String) (ClassLoader.java:742)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:362)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:367)
        at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
        at java.lang.Class java.lang.Class.classForName!(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:-2)
        at java.lang.Class java.lang.Class.forName(java.lang.String, boolean, java.lang.ClassLoader) (Class.java:400)
        at java.util.List org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(java.util.Collection) (ClassManager.java:214)
        at void org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses() (ClassManager.java:257)
        at void com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses() (ClassManagerFactory.java:94)
        at void org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(android.os.Bundle) (FtcRobotControllerActivity.java:356)
        at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:6709)
        at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1118)
        at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:2619)
        at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:2727)
        at void android.app.ActivityThread.-wrap12(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
        at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1478)
        at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:102)
        at void android.os.Looper.loop() (Looper.java:154)
        at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6121)
        at java.lang.Object java.lang.reflect.Method.invoke!(java.lang.Object, java.lang.Object[]) (Method.java:-2)
        at void com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run() (ZygoteInit.java:905)
        at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:795)
W/ClassManager: com.qualcomm.robotcore.wifi.WifiDirectAssistant$1 java.lang.ClassNotFoundException: com.qualcomm.robotcore.wifi.WifiDirectAssistant$1
E/ClassManager: java.lang.ClassNotFoundException: com.qualcomm.robotcore.wifi.WifiDirectAssistant$1
E/ClassManager:     at java.lang.Class.classForName(Native Method)
E/ClassManager:     at java.lang.Class.forName(Class.java:400)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(ClassManager.java:214)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses(ClassManager.java:257)
E/ClassManager:     at com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses(ClassManagerFactory.java:94)
E/ClassManager:     at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:356)
E/ClassManager:     at android.app.Activity.performCreate(Activity.java:6709)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
E/ClassManager:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619)
E/ClassManager:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
[... 141 lines cut ...]
E/ClassManager: java.lang.ClassNotFoundException: org.ejml.dense.row.FMatrixComponent
        at java.lang.Class.classForName(Native Method)
E/ClassManager:     at java.lang.Class.forName(Class.java:400)
[... cut ...]
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795)
E/ClassManager: Caused by: java.lang.ClassNotFoundException: Didn't find class "org.ejml.dense.row.FMatrixComponent" on path: DexPathList[[],nativeLibraryDirectories=[/system/lib64, /vendor/lib64]]
E/ClassManager:     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
E/ClassManager: 	... 18 more
```

Windwoes, 2021-11-24:
I've seen this behavior before, but the stacktrace you've posted is unrelated. That is "normal" on any RC app startup. (Not sure why, but it is).

When I saw this last, this is what caught my eye:
[image]

mihir-chauhan, 2021-11-30:
Yes, this is probably the real issue causing the crash, I was unsure which of the errors was causing it so I just added that stacktrace as well. If this is normal, how can I still debug the code through ADB as I am wirelessly connected to my hub? Thanks

Windwoes, 2021-11-30:
I'm not sure what causes the crash after deploying, and I've only seen it happen a couple times. I'd probably suggest clearing the Android Studio and Gradle caches and see if that does anything.

MrMattS, 2023-02-03:
Did you figure out a solution?