FIRST-Tech-Challenge/FtcRobotController #241: Use GitHub Templates

ryleu, 2021-11-24:
Howdy! I noticed that y'all get a ton of random / accidental PRs, so I wanted to let y'all know about GitHub's template feature. It works similarly to forking, except it prompts you to make a new repo name and it doesn't have an option to PR upstream.

Here's an example of what that would look like: https://github.com/FabricMC/fabric-example-mod

cmacfarl, 2021-11-25:
Chuckle.  Yeah.  There's been discussion surrounding a GitHub action that would automatically close pull requests.  It just hasn't yet been a real high priority.  For one, I don't really mind closing pull requests, and two when someone opens a PR against FtcRobotController and then realizes "oops", that's a learning experience in a relatively safe environment for that student.  I don't necessarily see that as a bad thing.  Combine those two perspectives and /shrug.

But the next release, or maybe the release after that, who knows, will have an action that auto-closes pull requests with a graciously professional explanation of why.

ryleu, 2021-11-27:
> Chuckle.  Yeah.  There's been discussion surrounding a GitHub action that would automatically close pull requests.  It just hasn't yet been a real high priority.  For one, I don't really mind [...]

That seems like a great solution! I like the idea of explaining it to people so they can learn about it. I still think the templates could be beneficial though, because it prompts people to make a new repo name, so that can help them stay organized season to season. It's up to y'all.

ryleu, 2021-11-29:
I could also help write an action like that if y'all want. I have a lot of free time and a bit of experience with wrangling GH actions.

JJTech0130, 2022-01-04:
I think it would be useful to switch to templates, because currently forks are not searchable, which makes it a pain to find things in them.
Next time I will clone the repo, switch remotes, and push it to a separate repository, but that's complicated 😅

JJTech0130, 2022-01-04:
I have created a fork and marked it as a template here: https://github.com/JJTech0130/FtcRobotController

NoahAndrews, 2022-02-15:
The problem with templates is that creating a new repo from a template starts a brand new Git history, which means that you cannot update your SDK by pulling from upstream.

JJTech0130, 2022-02-15:
Hmm... yes didn't think of that.

closed 2022-07-25 by NoahAndrews