FIRST-Tech-Challenge/FtcRobotController #256: REV Blinkin Example code is too complicated

FTC14133, 2021-12-14:
As a relatively green programmer and someone who is teaching students new to programming, the REV Blinkin example code was really difficult to figure out. We were able to make a much more simple implementation of it [here](https://github.com/FTC14133/FTC14133-2021-2022/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/Subsystems/Lights.java), but if we are really aiming for students to learn programming we should be starting with simple applications of things first.

JJTech0130, 2022-01-07:
Here's the only important code in the whole thing:
```java
RevBlinkinLedDriver blinkinLedDriver = hardwareMap.get(RevBlinkinLedDriver.class, "LED");
blinkinLedDriver.setPattern(RevBlinkinLedDriver.BlinkinPattern.GOLD);
```
Just replace `RevBlinkinLedDriver.BlinkinPattern.GOLD` with whatever pattern you want to display, `RevBlinkinLedDriver.BlinkinPattern.COLOR_WAVES_RAINBOW_PALETTE`, etc.

JJTech0130, 2022-01-07:
Oh I misread that, sorry. I thought you were asking how and linking to the existing example. 🤦‍♂️

CalKestis, 2022-11-22:
The simple example shown here is simple.  The goal with the existing sample was to show how one could change light patterns.  There are no plans to change the existing sample.

closed 2022-11-22 by CalKestis