FIRST-Tech-Challenge/FtcRobotController #271: JCenter Error

tom-ricci, 2022-01-12:
This SDK requires packages from JCenter. At the time of writing, https://jcenter.bintray.com/ seems to be down, and is returning a 502. While JCenter does claim it's a temporary outage, I wouldn't be surprised if it's permanently down considering JCenter is deprecated. 

If it goes back up, ignore this issue, I just decided to make it considering if JCenter is retired for good FIRST should probably switch to another dependency sooner than later.

cmacfarl, 2022-01-12:
With version 6.2.0 of the SDK artifacts were migrated to Maven.   Please update your SDK.

The latest version, v7.0, may be found here:  https://github.com/FIRST-Tech-Challenge/FtcRobotController

closed 2022-01-12 by cmacfarl

tom-ricci, 2022-01-12:
My bad! I just found out my team added Kotlin support with libs from JCenter last season and I assumed they were added officially since some of the official SDK Gradle files contain `jcenter()` in them.