FIRST-Tech-Challenge/FtcRobotController #280: Driver Hub not displaying ping times

jdkunzeman, 2022-01-25:
As an FTA at a recent meet, I noticed that some teams' driver hubs were not showing ping times prior to the match.  Some had something like ".Link...63..Hz" in the field where ping time is usually shown and one had some type of "not found" or "N/A" message where the ping time and channel are usually shown.  So far, I haven't found anywhere this "issue" has been reported.  I say "issue" because it did not seem to have any impact on robot operation.  Having the correct ping time shown is one way to evaluate the network.

Windwoes, 2022-01-25:
Is it possible that the area where the ping time is displayed was tapped? Tapping that region will switch between displaying the ping/channel and linkspeed/rxdbm.

jdkunzeman, 2022-01-25:
@Windwoes that's entirely possible.  I did not know that the display could be switched like that.  Wondering where the documentation is and how to interpret/use the linkspeed/rxdbm information.  I don't have a driver hub with me.  Will investigate more fully next time I can get my hands on one, then possibly close the issue.

Windwoes, 2022-01-25:
Yeah, it's only a feature on the landscape UI.
I think the only documentation that really mentions it is the release notes of the SDK to be honest.

Basically rx dBm gives the received signal strength of the access point, and link speed indicates how fast the link can theoretically transmit data. Those two metrics together are what determines how many signal bars are shown. You could have a high signal strength, but low link speed due to a noisy environment or such, or in a clean environment you might have relatively weak signal strength but still a fairly fast link speed.

Windwoes, 2022-01-25:
The "radio stats N/A" shows up when using a phone RC with the Driver Hub, because the WiFi Direct API doesn't support querying those statistics.

Also, the Driver Hub OS has a bug which seems to "freeze" the link speed metric immediately after connecting to the access point...

jdkunzeman, 2022-01-27:
@Windwoes thanks for the information.  Will close this issue since I verified the tap to change feature.  Would be good to document somewhere how to use the link speed/rx dBm info for troubleshooting network problems, if it can be used for that.  This is possibly a good topic for a slide during the next FTA/CSA forum.

closed 2022-01-27 by jdkunzeman