FIRST-Tech-Challenge/FtcRobotController #305: Add hprof files to .gitignore

abidingabi, 2022-03-13:
Gradle generates files like `java_pid17860.hprof` when it runs out of memory when working. These are dumped in the root of the project folder, and can be hundreds of megabytes in size. I have relatively commonly seen people having issues with these files and version control, as they accidentally commit these files and this causes issues (namely, GitHub does not allow files over 100 MB). A simple way to make this issue go away in terms of version control would be to add `*.hprof` to the .gitignore file of the SDK.

abidingabi, 2022-03-13:
In addition, increasing the maximum memory that is allocated to gradle (e.g. `org.gradle.jvmargs=-Xmx2048m` in `gradle.properties`) could help prevent the hprof files from being generated as often in the first place.

Ely31, 2023-09-07:
.hprof was added to the gitignore in v8.2, so this issue should be closed.

closed 2023-09-07 by abidingabi