FIRST-Tech-Challenge/FtcRobotController #307: Request to adjust Dualshock controller light strip through the SDK?

alan412, 2022-03-16:
(This is related to #306 )

While rumble is a better way for giving feedback, letting the robot code control the controller light strip would allow for a little more fun for the students as well as neat things for outreach.   (Also I like the idea of letting the students learn that rumble is a better way for giving feedback rather than it being the only way.)

Just my $.02

AndrewDTR, 2022-03-16:
I haven't even thought of the outreach possibilities -- would definitely be a fun idea to implement

Windwoes, 2022-07-25:
Implemented in v7.2.

closed 2022-07-25 by Windwoes

alan412, 2022-07-25:
@Windwoes: I just went and looked how it is implemented and wanted to give my appreciation for the way this was done!!  I really like the LedEffects and the builder. 👏

Windwoes, 2022-07-25:
> @Windwoes: I just went and looked how it is implemented and wanted to give my appreciation for the way this was done!! I really like the LedEffects and the builder. clap

Glad to hear it.