FIRST-Tech-Challenge/FtcRobotController #325: Unable to build project using gradle from command line

AlexD70, 2022-07-25:
Note that I am using Fedora Linux and do not want to use Android Studio because of bloat. When I first tried to use the gradlew script, I got an error with groovy's InvokerHelper, but after fixing the gradlew properies to use the source url of the 7.5 gradle version, but something is still wrong, giving an error that says it couldn't instantiate java's Runnable. It also says deprecated methods were used that are going to be removed in gradle 8. In addition, there is very little documentation on how this sdk actually works, making it difficult to write my own gradle build files.

cmacfarl, 2022-07-25:
Where does your ANDROID_HOME point?

AlexD70, 2022-07-25:
Honestly, I have no idea and rn i can't check, I have my android sdk in ~/Android so I think I can set that variable in ~/.bashrc.

cmacfarl, 2022-07-25:
v7.2 requires Android Studio Chipmuck 2021.2.1 or above.

I've verified that a gradlew assemble completes successfully in my workspace, on which laptop I have upgraded to Chipmunk.  My guess is that the Android SDK you are building against is incompatible.

AlexD70, 2022-07-25:
I set ANDROID_HOME to the right path and after updating the repo from the main branch stuff seems to be working. I also checked my java version to be sure it isn't old (it is the newest in fedora's repos. The main problem was that I was using an older repo with the newest gradle.

closed 2022-07-25 by AlexD70