FIRST-Tech-Challenge/FtcRobotController #340: No program is running

Q-TechFTC, 2022-08-28:
When I try to build and run a program to the driver station, It always returns in these two errors:
`E/ClassManager: java.lang.ClassNotFoundException: org.firstinspires.ftc.teamcode.NewT_proto_2$ToggleButton
E/ClassManager:     at java.lang.Class.classForName(Native Method)
E/ClassManager:     at java.lang.Class.forName(Class.java:324)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(ClassManager.java:215)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses(ClassManager.java:258)
E/ClassManager:     at com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses(ClassManagerFactory.java:94)
E/ClassManager:     at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:356)
E/ClassManager:     at android.app.Activity.performCreate(Activity.java:6251)
E/ClassManager:     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1107)
E/ClassManager:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2369)
E/ClassManager:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
E/ClassManager:     at android.app.ActivityThread.-wrap11(ActivityThread.java)
E/ClassManager:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
E/ClassManager:     at android.os.Handler.dispatchMessage(Handler.java:102)
E/ClassManager:     at android.os.Looper.loop(Looper.java:148)
E/ClassManager:     at android.app.ActivityThread.main(ActivityThread.java:5417)
E/ClassManager:     at java.lang.reflect.Method.invoke(Native Method)
E/ClassManager:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
E/ClassManager:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
E/ClassManager: Caused by: java.lang.ClassNotFoundException: Didn't find class "org.firstinspires.ftc.teamcode.NewT_proto_2$ToggleButton" on path: DexPathList[[],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
E/ClassManager:     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
E/ClassManager: 	... 18 more
E/ClassManager: 	Suppressed: java.lang.ClassNotFoundException: Didn't find class "org.firstinspires.ftc.teamcode.NewT_proto_2$ToggleButton" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]
E/ClassManager:     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
E/ClassManager: 		... 19 more
E/ClassManager: 		Suppressed: java.lang.ClassNotFoundException: org.firstinspires.ftc.teamcode.NewT_proto_2$ToggleButton
E/ClassManager:     at java.lang.Class.classForName(Native Method)
E/ClassManager:     at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
E/ClassManager:     at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
E/ClassManager: 			... 20 more
E/ClassManager: 		Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
W/ClassManager: org.firstinspires.ftc.teamcode.IMU_telemetry$2 java.lang.ClassNotFoundException: org.firstinspires.ftc.teamcode.IMU_telemetry$2
E/ClassManager: java.lang.ClassNotFoundException: org.firstinspires.ftc.teamcode.IMU_telemetry$2
E/ClassManager:     at java.lang.Class.classForName(Native Method)
E/ClassManager:     at java.lang.Class.forName(Class.java:324)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(ClassManager.java:215)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses(ClassManager.java:258)
E/ClassManager:     at com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses(ClassManagerFactory.java:94)
E/ClassManager:     at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:356)
E/ClassManager:     at android.app.Activity.performCreate(Activity.java:6251)
E/ClassManager:     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1107)
E/ClassManager:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2369)
E/ClassManager:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
E/ClassManager:     at android.app.ActivityThread.-wrap11(ActivityThread.java)
E/ClassManager:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
E/ClassManager:     at android.os.Handler.dispatchMessage(Handler.java:102)
E/ClassManager:     at android.os.Looper.loop(Looper.java:148)
E/ClassManager:     at android.app.ActivityThread.main(ActivityThread.java:5417)
E/ClassManager:     at java.lang.reflect.Method.invoke(Native Method)
E/ClassManager:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
E/ClassManager:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
E/ClassManager: Caused by: java.lang.ClassNotFoundException: Didn't find class "org.firstinspires.ftc.teamcode.IMU_telemetry$2" on path: DexPathList[[],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
E/ClassManager:     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
E/ClassManager: 	... 18 more
E/ClassManager: 	Suppressed: java.lang.ClassNotFoundException: Didn't find class "org.firstinspires.ftc.teamcode.IMU_telemetry$2" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]
E/ClassManager:     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
E/ClassManager: 		... 19 more
E/ClassManager: 		Suppressed: java.lang.ClassNotFoundException: org.firstinspires.ftc.teamcode.IMU_telemetry$2
E/ClassManager:     at java.lang.Class.classForName(Native Method)
E/ClassManager:     at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
[... 2 lines cut ...]
E/ClassManager: 			... 20 more
E/ClassManager: 		Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available`

`E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qualcomm.ftcrobotcontroller, PID: 9074
    android.view.WindowManager$BadTokenException: Unable to add window -- token null is not for an application
        at android.view.ViewRootImpl.setView(ViewRootImpl.java:571)
        at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:310)
        at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:85)
        at android.app.Dialog.show(Dialog.java:319)
        at android.app.AlertDialog$Builder.show(AlertDialog.java:1112)
        at com.qualcomm.ftccommon.FtcRobotControllerService.showWifiDirectConnectedAsPeerDialog(FtcRobotControllerService.java:586)
        at com.qualcomm.ftccommon.FtcRobotControllerService.onNetworkConnectionEvent(FtcRobotControllerService.java:538)
        at com.qualcomm.robotcore.wifi.WifiDirectAssistant.sendEvent(WifiDirectAssistant.java:658)
        at com.qualcomm.robotcore.wifi.WifiDirectAssistant$WifiDirectConnectionInfoListener.onConnectionInfoAvailable(WifiDirectAssistant.java:171)
        at android.net.wifi.p2p.WifiP2pManager$Channel$P2pHandler.handleMessage(WifiP2pManager.java:760)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
`

These errors appear even without my simple program to make the robot move, so it is not my code, but something with my settings or this package. Additionally, when I tried to search these errors up, people said it had something to do with instant run, which no longer exists as of Android Chipmunk.

Windwoes, 2022-08-28:
Can you please format the stacktrace output as a code block using the triple backticks?

AlecHub, 2022-08-29:
> When I try to build and run a program to the driver station ...

Hi @Q-TechFTC, what type of devices are you using for Driver Station and Robot Controller?  Phones, Driver Hub, Control Hub, Expansion Hub?

What do you mean by "run a program to the driver station".  Your ftcrobotcontroller APK should be running on a Robot Controller device.

Q-TechFTC, 2022-08-29:
Thank You! @AlecHub  We thought we had to run the program on the driver station since it seemed unwieldy to run the program on the robot controller(we are a rookie team). It works now.

closed 2022-08-29 by Q-TechFTC