FIRST-Tech-Challenge/FtcRobotController #343: Update with Power Play TF targets

REM-Codes, 2022-09-11:
Please update the examples to use the Power Play (2022-2023) signal targets (the three unique images). Also please release the 2023 official season release (currently the latest release is 2022 offseason)

Windwoes, 2022-09-11:
The 8.0 SDK is slated for release soon. Historically, SDK release has been anywhere from a few days after kickoff to a couple weeks.

kevinfrei, 2022-09-12:
Will the 8.0 SDK just be an update to this repo? That sure would make getting our programmers started easier. Rebasing on the 8.0 SDK is much easier than cross-repo cherry picking :/

NoahAndrews, 2022-09-12:
> Will the 8.0 SDK just be an update to this repo? That sure would make getting our programmers started easier. Rebasing on the 8.0 SDK is much easier than cross-repo cherry picking :/

Yes, barring some kind of significant change in programming methodology, you can expect that new versions will always be pushed to this repo, with a consistent git history.

closed 2022-09-14 by Windwoes