FIRST-Tech-Challenge/FtcRobotController #373: Robot Controller crashes when attempting to use Everyday Object TensorFlow model

j5155, 2022-09-23:
Attempting to follow the steps in [this](https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Using-a-TensorFlow-Pretrained-Model-to-Detect-Everyday-Objects) wiki article causes the robot controller app to crash with no indication of an error on the driver station; the only way to see that the app has crashed is viewing the Control Hub's internal screen via scrcpy and this error reported in logcat.
Either the Vuforia version should be changed to allow use of this model, or this error should be reported in an easier to diagnose way and the wiki article should be updated with a different model.
Logcat error:
`2022-09-22 16:53:14.269 2971-3152/com.qualcomm.ftcrobotcontroller E/AndroidRuntime: FATAL EXCEPTION: VuforiaFrameGenerator
    Process: com.qualcomm.ftcrobotcontroller, PID: 2971
    java.lang.AssertionError: Error occurred when initializing ObjectDetector: Object detection models require TFLite Model Metadata but none was found
        at org.tensorflow.lite.task.vision.detector.ObjectDetector.initJniWithByteBuffer(Native Method)
        at org.tensorflow.lite.task.vision.detector.ObjectDetector.access$100(ObjectDetector.java:86)
        at org.tensorflow.lite.task.vision.detector.ObjectDetector$3.createHandle(ObjectDetector.java:211)
        at org.tensorflow.lite.task.core.TaskJniUtils.createHandleFromLibrary(TaskJniUtils.java:91)
        at org.tensorflow.lite.task.vision.detector.ObjectDetector.createFromBufferAndOptions(ObjectDetector.java:207)
        at org.firstinspires.ftc.robotcore.internal.tfod.TfodFrameManager2$RecognizerPipeline2.<init>(TfodFrameManager2.java:66)
        at org.firstinspires.ftc.robotcore.internal.tfod.TfodFrameManager2$RecognizerPipeline2.<init>(TfodFrameManager2.java:55)
        at org.firstinspires.ftc.robotcore.internal.tfod.TfodFrameManager2.createRecognizerPipeline(TfodFrameManager2.java:52)
        at org.firstinspires.ftc.robotcore.internal.tfod.TfodFrameManager$MainPipeline.init(TfodFrameManager.java:227)
        at org.firstinspires.ftc.robotcore.internal.tfod.VuforiaFrameGenerator.run(VuforiaFrameGenerator.java:147)
        at java.lang.Thread.run(Thread.java:761)`

lizlooney, 2022-09-23:
@j5155 
In this step of the wiki: https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Using-a-TensorFlow-Pretrained-Model-to-Detect-Everyday-Objects#load-model-from-a-file

Did you remember to remove the following line?
```
tfodParameters.isModelTensorFlow2 = true;
```

j5155, 2022-09-23:
I did not, which is likely my error so I'm closing this issue for now; will test on the robot on Monday to confirm. Thanks for the quick response!

closed 2022-09-23 by j5155

lizlooney, 2022-09-23:
Thanks for your quick response! I think we'll update the wiki to explicitly list the changes that need to be made in that step.

lizlooney, 2022-09-23:
I updated the wiki. Thanks for the bug report!