FIRST-Tech-Challenge/FtcRobotController #378: update readme

PatrickSedlacek, 2022-09-24:
Readme has out of date season info. Still lists freight frenzy. To avoid confusiom list Power Play so teams know they have the correct SDK

AlecHub, 2022-09-26:
Hello Patrick.  I feel you've raised a valid issue.  Furthermore, the repo contains the Power Play SDK, but also contains the Freight Frenzy SDK and the Ultimate Goal SDK.  So mentioning only the Power Play SDK in the "NOTICE" section of the Readme is likely to cause confusion for people looking for the SDK of another season.  For example, people will be looking for the Freight Frenzy SDK to support the Modern Robotics electronics on their outreach bots and/or trainer bots.  Perhaps the "NOTICE" section should be eliminated.  This will eliminate maintenance of the section in the Readme.  People can look to the release notes for the season info.

holo-lb, 2022-09-30:
Bumped for clarity.
I suggest a change to the README from:
`This repository contains the public FTC SDK for the Freight Frenzy (2021-2022) competition season.`
to:
`The latest v8.0 release of this repository contains the public FTC SDK for the POWERPLAY (2022-2023) competition season.`

texasdiaz, 2022-09-30:
There is no need to "bump" this. No changes can be made to this public repository until a new export of the FTC SDK is released, due to the nature of the export process no files can or should be modified individually after the release process has taken place. There is absolutely no reason to force an 8.0.1 release to teams over a silly omission in the readme. This issue serves as notice that the README is incorrect and needs update. There is no more that needs to be done until a new export is ready to be released.

AlecHub, 2022-09-30:
> ... No changes can be made to this public repository until a new export of the FTC SDK is released, due to the nature of the export process no files can or should be modified individually after the [...]

There are several issues opened and closed on these questions.  These are recurring duplicate questions and issues.  Fixing the repo's [Contributing](../blob/master/.github/CONTRIBUTING.md#contributing-to-the-ftc-sdk) page so that it is clearer on these questions should put an end to the duplicate issues.

PatrickSedlacek, 2022-09-30:
Yep, After this was rejected I was going to suggest changing the contributing page

cmacfarl, 2022-09-30:
We are taking another look at the contributing page and there will be some changes, however the contributing page at present does say, effectively, don't issue a pull request against the upstream repository.   And it does say explicitly, quote, "Please note however that the deployment model of the SDK does not support direct pulls into ftctechnh/ftc_app."  Yeah, yeah, that's both the wrong account and repo, and yeah that will be fixed also in v8.1, but you get the picture.

The problem however is twofold, the Contributing Guidelines url is extremely small and below the text input box on the Pull Request form, and that leads to the second problem, that it's not clear how many people read and understand the Contributing Guidelines.  

We could link to the guidelines in the pull request template, which was probably an oversight not to do originally, although it's not clear by what percentage that would raise the readership, or we could just dump the guidelines into the pull request template, but that is a rather large annoyance for everyone who is using pull requests for merges between branches in their forks by virtue of forcing them to delete a wall of text before adding their own description intended for consumption by other team members.  Or we could be more explicit in the pull request template although I don't know exactly what the wording should be to prevent team members from inadvertently opening a PR back upstream vs. opening a PR against the local fork.   We don't necessarily want to discourage the latter.

alan412, 2022-09-30:
I try to encourage teams to add the FTC repo as an "upstream" repo instead
of forking it which gets rid of the problem of people issuing a pull
request to the wrong repo.   I have had a request to add an appendix to the
book Learn Java for FTC ( https://github.com/alan412/LearnJavaForFTC )
 talking about how we setup git and Android Studio.

--Alan

AlecHub, 2022-10-01:
> We are taking another look at the contributing page and there will be some changes ...

Great!  Perhaps you can post your proposed changes here prior to releasing it in the next SDK.  That way we don't have to wait months and months for any fixes or improvements that need to be made to your changes.

> The problem however is twofold, the Contributing Guidelines url is extremely small and below the text input box on the Pull Request form, and that leads to the second problem, that it's not clear how many people read and understand the Contributing Guidelines ...

If you look below you will see the footnote "Remember, contributions to this repository should follow its [contributing guidelines](https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/.github/CONTRIBUTING.md)."  I believe the reason it is a footnote and the reason it says "Remember" is because GitHub users already know to read and understand a Repo's particular rules on contributing.  This is standard for GitHub repos.

If you prefer a more prominent place, then by all means put the contributing rules, the SDK processes, turnaround time for updates, etc., in the Readme.  I can't think of a single release where an issue wasn't created against the Readme, which leads me to believe that people are reading the Readme.

closed 2022-11-22 by CalKestis