FIRST-Tech-Challenge/FtcRobotController #383: Code cleanup for future FtcRobot controller

thoughtless42, 2022-09-27:
We have a script that takes care of these issues, but thought the larger ftc community could benefit from this request. 
1.)FIRST-Tech-Challenge/FtcRobotController/FtcRobotController/build.gradle
     compileOptions {
       sourceCompatibility JavaVersion.VERSION_1_7
       targetCompatibility JavaVersion.VERSION_1_7
    }

should  these be JavaVersion.VERSION_1_8 to match build.common.gradle ?

2.) FtcRobotController/FtcRobotController/src/main/assets/
this folder should be removed, it adds undue size to the robot controller apk and does not add value.
These historic files can be added to a new repository along with the pdf target files; split into separate dependencies like the last 2 seasons within build.dependencies.gradle
dependencies {
    implementation 'org.firstinspires.ftc:gameAssets-PowerPlay:1.0.0'
}

3.) FtcRobotController/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/internal/FtcOpModeRegister.java
remove import org.firstinspires.ftc.robotcontroller.external.samples.ConceptNullOp;
this import is not used  and puts an undefined dependency on the external/samplesfolder on FtcOpModeRegister.java

cmacfarl, 2022-09-30:
Thank you for reporting this.  It is being tracked internally.

cmacfarl, 2023-06-12:
> FtcRobotController/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/internal/FtcOpModeRegister.java
> remove import [...]

ConceptNullOp is actually being used in the javadoc with the @link property, which means it is required for the javadoc build so the resultant documentation can reference the class.

thoughtless42, 2023-06-12:
I see the reference on line 52, but i am not sure i follow why the import
needs to reference that particular opmode, with the additions of on the
block java and blocky for novice teams to leverage to the robot controller.

Ely31, 2023-09-07:
> 1.)FIRST-Tech-Challenge/FtcRobotController/FtcRobotController/build.gradle compileOptions { sourceCompatibility JavaVersion.VERSION_1_7 targetCompatibility JavaVersion.VERSION_1_7 }
> 
> should these be JavaVersion.VERSION_1_8 to match build.common.gradle ?
This was done in version 8.2.

Ely31, 2023-09-11:
the assets folder was removed in 9.0

qwertychouskie, 2023-10-30:
> if line 52 was * See, for example the class annotations in {@link org.firstinspires.ftc.robotcontroller.external.samples.ConceptNullOp}. then the import on line 37 import org.firstinspires.ftc.robotcontroller.external.samples.ConceptNullOp; should be able to be removed without breaking javadoc

Created https://github.com/FIRST-Tech-Challenge/FtcRobotController/pull/802 but it was auto-closed.

qwertychouskie, 2023-11-10:
@cmacfarl Just wanted to check in on the status of fixing the last part of this issue.  The fix already exists (https://github.com/FIRST-Tech-Challenge/FtcRobotController/pull/802) but due to the nature of SDK development, there's seemingly no way to properly submit these changes to upstream.  I also came across some more simple-but-potent optimizations in https://github.com/FIRST-Tech-Challenge/FtcRobotController/issues/797 but once again sadly PRs are not possible.