FIRST-Tech-Challenge/FtcRobotController #394: crash with custom TF2.x model

wangxdflight, 2022-10-05:
When I did my custom TF2.x model, tensor flow example app crashed when opmode was started.
"
initializing objectdetector: output tensor at index 0 is expected to have 3 dimensions, found 2.  "

Searched around and upgrading task vision version as below worked around the issue. Has anyone has similiar experiences?

-    implementation 'org.tensorflow:tensorflow-lite-task-vision:0.2.0'
+    implementation 'org.tensorflow:tensorflow-lite-task-vision:0.3.1'

texasdiaz, 2022-10-05:
What tool did you use to create your custom model?

texasdiaz, 2022-10-06:
It seems as if multiple people in the past had this issue with models built with TensorFlow version 2.6 using the same basic template with CoLab. It appears that the template has an error, which inverts the indices incorrectly. This link may help:

https://github.com/tensorflow/tensorflow/issues/51591#issuecomment-923051299

wangxdflight, 2022-10-06:
I did went over those workarounds, but no luck. Reverting to TF2.5 caused many PIP dependency issues.
I tested with tensorflow-lite-task-vision:0.3.1, it cannot detect at all.
You are right. It is hard to use Colab templates found online.

closed 2022-10-07 by wangxdflight