FIRST-Tech-Challenge/FtcRobotController #424: Make v7.1 SDK Open Source

AlecHub, 2022-10-21:
Our outreach bots use [Modern Robotics Core Control Modules](https://modernroboticsinc.com/product-category/core-control-modules/).  Therefore, we are using v7.1 SDK to build the Robot Controller App and using v7.1 Driver Station App.  

However, the Robot Controller and Driver Station Apps put out the warning: 
```diff
! The Robot Controller and Driver Station apps are both obsolete. You should install the new versions for this FTC season.
```
Request to make v7.1 Open Source so that the community can address issues and implement enhancements.

texasdiaz, 2022-10-21:
Greetings, Alec!

1. Teams can disable the obsolete software warning from within the Advanced RC Settings menu.
2. [OpenRC](https://github.com/OpenRC/openrc) is a good resource for those who would like to make custom SDK software changes for offseason/demo use.

-Danny

NoahAndrews, 2022-10-21:
To follow on to @texasdiaz's point, Extracted-RC has a branch for each SDK release. [Here's the one for 7.1](https://github.com/OpenFTC/Extracted-RC/tree/7.1). I think it would be really cool to see a community project start up that backports selected changes from 7.2 and later to 7.1. If someone were to start that project, I'd probably turn on notifications for it, and would be happy to provide some backporting guidance (time-permitting).

For anyone who's not familiar with Extracted-RC, the Robot Controller app is actually already open-source. The source code is just distributed in source JAR files. Extracted-RC just takes the source code from those JAR files and puts it into an easy-to-modify project that will build the RC app.

---

Speaking as someone who has contributed a lot of code to the official SDK, most changes only require modifying the Robot Controller app, and will work just fine with a older Driver Station.

AlecHub, 2022-10-22:
Thank you @texasdiaz and @NoahAndrews !!!

closed 2022-10-22 by texasdiaz