FIRST-Tech-Challenge/FtcRobotController #437: setZoom for USB camera

wangxdflight, 2022-10-29:
I tried to use PtzControl.setZoom, but no effect on my Logitec C920 USB camera.

setZoom works well on Android internal camera without problem. Wondering if it is a limitation with UVC camera driver.

Any help here? Thanks!

WestsideRobotics, 2022-10-29:
That function does work on that camera, as described here:
http://ftc-docs.firstinspires.org/programming_resources/vision/webcam_controls/webcam-controls.html#zoom

Suggest you read the full article, and see the sample OpModes provided.

wangxdflight, 2022-10-29:
Thanks...I did see the minZoon is 100 and maxZoom is 500. But didn't realize **" although values higher than 250-280 have no further effect on the preview image (influenced by Vuforia)."**

Good information. Let me try it.

closed 2022-10-29 by wangxdflight