FIRST-Tech-Challenge/FtcRobotController #466: Kotlin Compilation Errors

lukasmwerner, 2022-11-10:
Some of the members of my team would like to utilize some kotlin abilities. However, whenever a kotlin file is added the gradle compilation breaks.

Are there additional changes that need to be completed to add kotlin on a Control Hub?

texasdiaz, 2022-11-10:
Please understand that while FTC does not prevent teams from using Kotlin, FTC does not provide technical support for Kotlin within the FTC control system. Perhaps another user might be able to provide some support if they've been able to get it to work.

lukasmwerner, 2022-11-10:
While I understand that FTC would not like to provide support I would like to point out that Google has been ["Kotlin First"](https://developer.android.com/kotlin/first) on the android platform.

lukasmwerner, 2022-11-10:
Closing this as I managed to find the right kotlin version to install. Add the following to the TeamCode `build.gradle` file

```gradle
implementation 'androidx.core:core-ktx:1.6.0'
```

This should get the versions to match with the REV Control Hub's OS Version

closed 2022-11-10 by lukasmwerner

ofluffydev, 2024-09-27:
Had to use this solution after adding road runner, thanks!

AlecHub, 2024-09-27:
I've never used Kotlin.  I would love to see a comparison of Kotlin and Java.  For example, initializing and calibrating the IMU in a Kotlin Coroutine vs a Java (or Kotlin?) Thread.

lukasmwerner, 2024-09-27:
We ended up open sourcing our entire team code [here](https://github.com/LaSalleRobots/Freight-Frenzy)

ofluffydev, 2024-09-27:
It's supposed to be "safer" with null values and easier to learn

https://kotlinlang.org/docs/comparison-to-java.html

lukasmwerner, 2024-09-30:
You can learn a lot from their excellent [documentation here](https://kotlinlang.org/docs/coroutines-basics.html#scope-builder-and-concurrency)

From what I can tell you should just be able to write something like this:

```kotlin
import kotlinx.coroutines.*

// Sequentially executes doWorld followed by "Done"
fun main() = runBlocking {
    doWorld()
    println("Done")
}

// Concurrently executes both sections
suspend fun doWorld() = coroutineScope { // this: CoroutineScope
    // these two launch are running in the "background"
    launch {
        delay(2000L)
        println("World 2")
    }
    launch {
        delay(2000L)
        println("World 1")
    }
    println("Hello")
}
```

I won't be able to help with any actual code because I've been an FTC alumni for 3 years now and no longer have any hardware to test this stuff on.