FIRST-Tech-Challenge/FtcRobotController #469: Broken Everyday Object Detection example?

lalondma, 2022-11-13:
Hi, I'm trying to play with the pretrained tflite model for "Everyday Object detection" using the given source code (commit https://github.com/FIRST-Tech-Challenge/FtcRobotController/wiki/Using-a-TensorFlow-Pretrained-Model-to-Detect-Everyday-Objects/1e8896b68d41bbfaf2954bb0dd6319a63b852928), the OpMode is running without error but detection per se is broken:
 
[image]
[image]

This usually happens to me when there's some mismatch between training and inference source code w.r.t image preprocessing (e.g. color normalization). I haven't changed the code except for camera selection (webcam) and lower detection threshold... Could the problem be related to the camera (e.g. different pixel color format)? 
Robot controller version 8.0
Control hub OS version 1.1.3 (firmware 1.8.2)

texasdiaz, 2022-11-13:
It's not broken at all. Depending on the neutral network type, a neutral network may actually have hundreds of detections within an image, with the vast majority of them being low confidence detections. Tensorflow uses the minimum confidence threshold to filter out low confidence detections, and it also uses other algorithms to remove duplicate detections in the same area. We also don't return ALL of the detections from the network. If you want the higher detection confidence, which hopefully provides a better detection result, you need to set the minimum confidence higher.

-Danny

lalondma, 2022-11-13:
It's obvious that I'll get more detections (and most likely garbage) if I lower the confidence threshold, and I know about the removal of duplicate detections (NMS), but something else is happening: 'easy' COCO objects like laptop, keyboard, cup, stop sign, etc. aren't detected even when the confidence threshold is low, and unsurprisingly it's worse if I raise the threshold (no detections at all). Is there a way to have access to the tensorflow pipeline? I'd check the input tensor colorspace and mean values...

texasdiaz, 2022-11-13:
No, there's no way to get access to the tensorflow pipeline.

You *are* adding ALL of the labels from the label map, right? I am assuming you are, but that's a common error that people make.

You might also want to try out using the TensorFlow 2.x COCO models that we have available for you to use. You can find the link to the Model Zoo models we converted here:
https://ftc-community.firstinspires.org/t/tensorflow-model-zoo-models/159/2

I mention this just in case something odd is happening with your zoo model. I personally use a C270 camera on a Control Hub using the TensorFlow 2.x Model Zoo model, and I get great results. Make sure you don't have camera calibration files being applied nor any other features that may be changing how TensorFlow is seeing the camera image.

-Danny

lalondma, 2022-11-13:
Yes, I'm using the whole labelmap file. I've tested your TF2.x models and now I get good detections with decent (>0.7) confidence score. Oddly, labels are shifted by one position, i.e. I get 'mouse' instead of 'laptop', 'parking meter' instead of 'stop sign', etc. Adding a line to the labelmap file fixed the issue. Maybe I'll test the initial TF1.x model on a Raspberry PI, just out of curiosity...

texasdiaz, 2022-11-14:
I bet if you compared the label map files you'd find they were slightly different (I haven't looked myself, but this is my hunch). While the two models might have the same origin, the conversion process was definitely different.

Glad the TF2 model gives you better results. I've not personally used the TF1 model, I can't vouch for its performance.

-Danny

lalondma, 2022-11-14:
I've compared the files, they are the same.  Seems that some people have had this label issue in the past: https://github.com/JerryKurata/TFlite-object-detection/blob/dd49053e1eaf31eb4e6c7da9e32905900c1f1d88/TFLite_detection_video.py#L86. Also, I've tested the TF1.x model on a Raspberry Pi and it worked fine. Maybe recent changes were made to the vuforia/tflite pipeline that impact performance for TF1.x models?

texasdiaz, 2022-11-17:
That's an interesting find, I have not used the TF 1.x models so I can't verify personally.

The only change this season to TensorFlow itself has been the upgrade in TensorFlow version, because the TensorFlow training network will not allow older versions of TensorFlow models to train soon. TensorFlow, being a research project, is fond of breaking old versions of things so it's very possible older models could be affected. 

We did make a change to allow other input sources (other than Vuforia) for images into TensorFlow in the SDK, but that change would not affect just TF 1.x models.

-Danny

baqwas, 2022-11-27:
@lalondma, thanks for the observation "_labels are shifted by one position_" That is exactly what our team kept reporting but neither they nor I had the debugging skills that you've displayed. Now that I know the issue lies with our label file (honestly, no changes were made and @texasdiaz had alerted us previously to strictly observe the sequence), I'll have to get a team member to revisit the label files. (The team members were confused precisely by the two specific mappings you mentioned. I should have taken a closer look at the txt file).

Kind regards.

texasdiaz, 2022-12-03:
Additional information that is likely helpful in investigating this issue:
[image]

texasdiaz, 2022-12-03:
I'm closing this topic, because I think the label maps are a red herring to the ultimate issue, which is described in #505 better.

closed 2022-12-03 by texasdiaz