FIRST-Tech-Challenge/FtcRobotController #474: Rookie teams can't use new Control Hubs

alan412, 2022-11-14:
The new Control Hubs have the new IMU: BHI260AP which is supposed to be supported in 8.1.    This is keeping rookie teams from being able to use the IMU.   (This is commonly used for field relative mecanum driving)

Please make available even a branch with that support early as scrimmages and competitions are coming.

Thanks,

Alan

worthp, 2022-11-18:
I just found the same thing. Is there an ETA on that driver.?

texasdiaz, 2022-11-19:
We are anticipating being able to release 8.1 early next week.

worthp, 2022-11-19:
Excellent, @texasdiaz!!! Thanks for the quick reply.

NoahAndrews, 2022-11-22:
8.1 is now live. See this tutorial for more details: https://ftc-docs.firstinspires.org/programming_resources/imu/imu.html

closed 2022-11-22 by NoahAndrews

worthp, 2022-11-22:
Whooo hooo!!! I"ll check it out tmrw and keep you posted.Thanks again

worthp, 2022-11-22:
@NoahAndrews i updated robot and drive station. new one is working as expected. thanks again for the quick turn and communication!