FIRST-Tech-Challenge/FtcRobotController #479: possible to connect REV control hub to WIFI router?

wangxdflight, 2022-11-17:
I understand that control hub comes up as a WIFI direct AP, but the WIFI performance is unsatisfactory. Can we change it to connect to WIFI router instead?

I tried to hook with a monitor to change the setup, but it seems its UI has been customized from normal android, there is no option for station mode.

Any help here? Thanks.

texasdiaz, 2022-11-17:
The Control Hub is meant to act as the central Access Point (AP) for the robot control system, to what all other devices (Driver Station, Programming Laptop, etc...) connects to. Teams are not allowed to have other Wi-Fi devices in competition (e.g. Routers, other APs or Hotspots). Hence, there is no means of converting the Control Hub into a Client (Station Mode) versus an Access Point.

Can I ask what metrics you are using to determine that the Wi-Fi performance is unsatisfactory?

-Danny

wangxdflight, 2022-11-17:
I meant for development use, not for competition.
Some team member's laptop have difficulty to connect to controller's WIFI, the connection drops frequently and connects to other wifi routers which might have stronger signal or for unknown reasons.
It might be a compatibility issue. Some other laptops don't have such issue. It might be laptop issues as well. But it seems wifi router connection is more stable.

texasdiaz, 2022-11-17:
Since the Control Hub does not have internet access, Windows networking puts a higher metric on Wi-Fi connections that do have internet access over that one. Therefore, if you have any other Wi-Fi connections marked as "Connect Automatically" or "Connect Automatically when in Range" (depending on your version of Windows) Windows will shift to the other Wi-Fi connections when the Control Hub reboots and temporarily disables the Wi-Fi. It's recommended when working with the Control Hub to uncheck the "Connect Automatically" option on all other remembered networks, and check "Connect Automatically" to the Control Hub SSID. 

We do not support swapping over the Control Hub to Station Mode.

-Danny

wangxdflight, 2022-11-17:
Thanks for the replies...

closed 2022-11-17 by wangxdflight