FIRST-Tech-Challenge/FtcRobotController #489: v8.1 file resolution errors

holo-lb, 2022-11-22:
Upon usage of the new release v8.1, Gradle was unable to build the app successfully due to missing dependancies on the server end. 
These files upon visiting their respective links report 404 Not Found.

This could be a simple release timing error that will fix very shortly, but just reporting this issue just in case. Thanks.

_Snippet of Gradle error message:_
```
FAILURE: Build completed with 9 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':TeamCode:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':TeamCode:debugRuntimeClasspath'.
   > Could not find org.firstinspires.ftc:Inspection:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/Inspection/8.1.0/Inspection-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/Inspection/8.1.0/Inspection-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Inspection-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Inspection.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:Blocks:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/Blocks/8.1.0/Blocks-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/Blocks/8.1.0/Blocks-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Blocks-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Blocks.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:Tfod:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/Tfod/8.1.0/Tfod-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/Tfod/8.1.0/Tfod-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Tfod-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Tfod.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:RobotCore:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/RobotCore/8.1.0/RobotCore-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/RobotCore/8.1.0/RobotCore-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/RobotCore-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/RobotCore.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:RobotServer:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/RobotServer/8.1.0/RobotServer-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/RobotServer/8.1.0/RobotServer-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/RobotServer-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/RobotServer.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:OnBotJava:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/OnBotJava/8.1.0/OnBotJava-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/OnBotJava/8.1.0/OnBotJava-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/OnBotJava-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/OnBotJava.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:Hardware:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/Hardware/8.1.0/Hardware-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/Hardware/8.1.0/Hardware-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Hardware-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/Hardware.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
   > Could not find org.firstinspires.ftc:FtcCommon:8.1.0.
     Searched in the following locations:
       - https://repo.maven.apache.org/maven2/org/firstinspires/ftc/FtcCommon/8.1.0/FtcCommon-8.1.0.pom
       - https://dl.google.com/dl/android/maven2/org/firstinspires/ftc/FtcCommon/8.1.0/FtcCommon-8.1.0.pom
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/FtcCommon-8.1.0.jar
       - file:/C:/Users/user/StudioProjects/BunyipsFTC/libs/FtcCommon.jar
     Required by:
         project :TeamCode
         project :TeamCode > project :FtcRobotController
```

CalKestis, 2022-11-22:
Oh my.  I'm an idiot.  I forgot to push the release button on the staged 8.1.0 repositories on sonatype/maven.

I did a sanity build after pushing 8.1 here and of course it was successful because gradle cached the 8.1 artifacts on my local machine and so I didn't realize that I forgotten to push the release button.

So sorry folks.  That should be fixed momentarily.

CalKestis, 2022-11-22:
Looks like it's there now, and an invalidate caches and rebuild was successful, which forces a pull of a new set of artifacts from maven, on my laptop

https://repo1.maven.org/maven2/org/firstinspires/ftc/RobotCore/8.1.0/

Again, sorry for the churn.

closed 2022-11-22 by CalKestis

frizelr4927, 2022-11-22:
Thanks, better late then never. 😊