FIRST-Tech-Challenge/FtcRobotController #492: Updated to SDK 8.1, can't create new blocks opmode?

dbwell00, 2022-11-26:
After update to 8.1, trying to create a new blocks opmode from any sample results in error "New project failed. Error code 0." I can upload existing .blk files, make copies of existing ones... just can't create new from template. Am I missing something

cmacfarl, 2022-11-26:
This is a known issue, which we have a fix for internally.  e.g. You'll likely see a 8.2 soon.

In the meantime, if you don't need 8.1 because you don't have a CH with the new IMU, or you simply aren't using the IMU you should revert back to 8.0.

texasdiaz, 2022-11-28:
@dbwell00 Would you like to help us test a release candidate patch for this issue?

texasdiaz, 2022-11-29:
There are two issues being addressed in the 8.1.1 patch, this blocks issue being the major issue. Please send an email to ftctech@firstinspires.org so I can send you instructions.

texasdiaz, 2022-12-03:
This has been fixed in 8.1.1.

closed 2022-12-03 by texasdiaz