FIRST-Tech-Challenge/FtcRobotController #495: 8.1 SDK: OpMode telemetry field is now final

kevinfrei, 2022-11-26:
Anyone using the FtcDashboard has to work around this. Anyone using libraries that try to integrate seamlessly with the Telemetry system probably has even more work to do.

Are there release notes somewhere that explain the logic behind this change? I'd be even more interested in reading the justification for deploying a breaking change (of *any* sort) mid-season.

kevinfrei, 2022-11-26:
(Ideally, could y'all undo the change for 8.2?)

texasdiaz, 2022-11-28:
@kevinfrei Would you be willing to help us test release candidate patch 8.1.1 once it is ready?

kevinfrei, 2022-11-28:
Absolutely!

texasdiaz, 2022-11-28:
Please send me an email to ftctech@firstinspires.org so I can notify you about the release candidate and provide instructions.

ymanchanda, 2022-11-28:
We use roadrunner and after updating to 8.1 we are getting a build error "cannot assign a value to final variable telemetry" in the opmodes, AutomaticFeedForwardTuner, DriveVelocityPIDTuner, ManualFeedForwardTuner for this line of code:
telemetry = new MultipleTelemetry(telemetry, dashboard.getTelemetry());

Please help!

cmacfarl, 2022-11-28:
@ymanchanda 

The problem is that your opmode, through its integration with FtcDashboard, wants to hijack the Telemetry object declared in OpModeInternal wherein it is declared with the 'final' modifier.  You can not assign a value to a member declared final outside of its initialization.  This was the result of some refactoring that appeared in v8.1.

We are in the process of turning the crank on a v8.1.1 release that will contain a fix for both this and https://github.com/FIRST-Tech-Challenge/FtcRobotController/issues/492

In the meantime, as a workaround, you should be able to declare a new Telemetry object with a name that's not 'telemetry' in your opmode.   Assign that the result of creating a new instance of MultipleTelemetry, and then use that new instance within your opmode, and things should work as before.  Disclaimer, I haven't actually done this hence all the normal "should work" disclaimers apply.

kevinfrei, 2022-12-02:
I fully understand the problem and the work around is obvious (at least it was to me...). Making something `final` is a breaking change, so the software industry veteran in me has concern about the level of discipline taken when modifying the SDK mid-season.

texasdiaz, 2022-12-03:
This has been fixed in 8.1.1.

closed 2022-12-03 by texasdiaz