FIRST-Tech-Challenge/FtcRobotController #501: TFOD inference time/latency?

wangxdflight, 2022-11-29:
We saw latency more than 1 second. Is it expected?
Possible to make it run faster?

Thanks.

texasdiaz, 2022-11-29:
You'll find that tweaking parameters for TFOD won't help our use-case - we don't/can't use multiple interpreters, our hardware can't handle it, and that's usually how you speed things up (throw images into more pipelines to do inference in parallel). The inference processing on the Control Hub or a legal FTC phone literally cannot get any faster - the only real thing you can do is enable tracking on the FTC API side to help track objects between inference detections; this way you can get ~15fps or more tracking once an object is identified/detected.

wangxdflight, 2022-11-29:
Thanks for the reply. We noticed that it was faster after an object was detected.
The processor on control hub is indeed not so powerful.

[image]

texasdiaz, 2022-11-29:
Yep, the challenge we face in defining the next "Revision" of the Control Hub is chip lifetimes, most processors for mobile devices have a lifespan of 1-2 years whereas we want 5+ years. So in our long-term roadmap planning we're looking at these kinds of things with an eye toward the future of FTC hardware!

closed 2022-11-30 by wangxdflight