FIRST-Tech-Challenge/FtcRobotController #505: Custom tensorflow object detection not working with SDK 8.*

Tarkkett, 2022-12-02:
Our team is trying to migrate to the newest SDK version, because it is required for the competition. The thing is, that earlier we had a fully working, custom tensorflow 1 object detection model and as soon as we updated the version, the model didn't work anymore (not detecting anything at all). Did the pre-trained model change? The same code works properly in SDK versions <7.2.(7.2 not included). We didn't find any breaking changes in the documentations of these newer versions :( Has anyone had this issue? If so, please let me know how to fix it.

texasdiaz, 2022-12-03:
This issue is a duplicate of #469.

NoahAndrews, 2022-12-03:
@texasdiaz This doesn't sound like a duplicate to me.

texasdiaz, 2022-12-03:
@NoahAndrews What's different? Both are claiming that the TensorFlow 1.x Everyday Objects models not "working" any more - in other words, TensorFlow's stock Model Zoo models from TF 1.x are not properly classifying detections any more. Both claim that something broke the 1.x models from "working." 

Did we break the 1.x model backwards compatibility when we separated Vuforia from TensorFlow? I don't know. Did a new version of TensorFlow libraries break backwards compatibility? I don't know. I'm not closing the issue because I don't know what the ultimate problem is. I just know both of these are effectively saying the exact same thing. The other issue came first, which is why I closed this one. This one has non-red-herring information in it. Maybe it makes sense to combine them.

The other issue confuses the topic by talking about the labels, if you're using the Model Zoo label map from 1.x there's an extra line in the label map (the first element is "????" in the 1.x map, but that extra line is not there in the 2.x map). I firmly believe that's a complete red herring. The "simple" and "easy" answer to both of these - until we figure out what went wrong - is to just use the TF 2.x Model Zoo models that Uday and I converted and provided for people to use. The more complex answer lies in determining what happened in SDK 7.2 that seems to have broken the TF 1.x compatibility.

-Danny

texasdiaz, 2022-12-03:
I reopened this topic because it now has a better title indicative of the actual problem, and is more succinct.

texasdiaz, 2022-12-03:
As a workaround for the time being, you should try using the TensorFlow 2.x COCO models that we have available for you to use. You can find the link to the Model Zoo models we converted here:

https://ftc-community.firstinspires.org/t/tensorflow-model-zoo-models/159/2

This only helps if the only TensorFlow 1.x model you want to use is the COCO Model Zoo (Everyday Objects) model.

lizlooney, 2022-12-05:
What is the value of the inputSize parameter when you initialize TensorFlow?

Tarkkett, 2022-12-06:
300 or 320, this depends on the model.

Tarkkett, 2022-12-06:
I just want to know which pre-trained model to use from the [Tensorflow model zoo](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/tf2_detection_zoo.md)

Tarkkett, 2022-12-06:
@texasdiaz I understand that TensorFlow Model Zoo models are full pre-trained tensorflow models and not .tflite. And the .tflite's you sent me are everyday object detection models, how are they useful for the competition? We are trying to take the ssd-mobilenet-v2 models from the model zoo(the .tflite's you sent me have the name ssd-mobilenet-v2 in them), train them, and exports as .tflite's. We did this successfully with TF1 models, but since TF1 is for some reason no longer supported, we are trying to use TF2. **I need to know for sure which pre-trained models you used and how did you convert them to .tflite.** Our .tflite models don't work, we get an error: "Unfortunatly, the robot controller has stopped", which gives us no information at all.

texasdiaz, 2022-12-06:
@Tarkkett I think I need to step back for a moment and ask what you're trying to do. How are you trying to train models? And what tool are you trying to train models with? 

When you take a TensorFlow Model Zoo model and train on top of them, you get the benefit of a fully formed neural network (meaning your training performance using the new data will be much better than trying to train a network from scratch) but you completely lose all of the labels from the original model because you were unable to train and evaluate using the dataset used to create the original Zoo model (and no, we do not have access to that dataset). 

So if you just use the "stock" zoo model you can use all of the trained labels and categories of the Zoo model as-is - meaning if you see an image of a dog the model will recognize it, if you see a keyboard the model will recognize it, and so on. HOWEVER, if you try to train on top of that model, you completely LOSE all of the Model Zoo labels, and you only get what you trained - and if you trained poorly, the model will be poor (garbage in, garbage out).

Does this make sense? Maybe you should tell me how you're TRYING to use the model zoo models, and we can go from there (rather than me assuming why you wanted the TF 1.x model zoo models to begin with).

lizlooney, 2022-12-06:
> Our .tflite models don't work, we get an error: "Unfortunatly, the robot controller has stopped", which gives us no information at all.

Can you send your .tflite file? I'd like to see if I can reproduce and debug the issue with it making the robot controller stop.

Tarkkett, 2022-12-07:
> > Our .tflite models don't work, we get an error: "Unfortunatly, the robot controller has stopped", which gives us no information at all.
> 
> Can you send your .tflite file? I'd like to see if I can reproduce and debug the issue with it making the robot controller stop.

You can download it [Here](https://we.tl/t-1tWTZFIR6l) , we made lots of models, but all of them have the same issue.

texasdiaz, 2022-12-07:
How did you make the model?

Tarkkett, 2022-12-07:
@texasdiaz We downloaded the pre-trained model from the zoo, then labeled many images and generated tfrecords from these files, we used our tfrecords and the fine_tune_checkpoint from the zoo to train the model, after training we got the saved_model. Converted that model to tflite_graph and after that, converted to tflite.

texasdiaz, 2022-12-07:
I see. Are you using TensorBoard to look at model metrics during training? Without using TensorBoard, you have no idea how well (or not) the model Training is going. Unfortunately one cannot simply "train" a model and then run it to see how well it does - that's just setting yourself up for frustration.

wangxdflight, 2022-12-07:
> 

Do you have crash call stack?  I had crash issue using my customized model, but was able to work around by using the right tflite version in generating model and android gradle file:
    implementation 'org.tensorflow:tensorflow-lite-task-vision:0.3.1

But your crash might be different though.

texasdiaz, 2022-12-07:
> @texasdiaz we are using tensorboard, the loss seems normal, but it should detect something even if the loss would be very high.

Not necessarily. I've had many models with high loss that didn't detect anything at all, because the network couldn't properly classify any object features. It came down to training. Are you doing like few-shot learning or anything to limit the number of training steps required, or how much training are you doing?

Tarkkett, 2022-12-07:
But it wasn't detecting anything at all, o it crashed? It would make sence if it wouldn't detect anything, because, there would be a very high detection count. I am not sure what few-shot learning is. We aren't training much, because now we don't care if the detections are messed up and garbage, we just want to load the model, without the hub crashing.

>

texasdiaz, 2022-12-07:
Oh, that's literally the first time you said "crash" - that's pretty important. Just not detecting anything vs crashing is a very different issue.

I would very much appreciate it if you could reproduce the crash, and then download your Robot Controller log. Copy-paste it into pastebin.com, and create a new paste. Then provide me with the url to your pastebin. That way I can look at your crash log to see what happened.

lizlooney, 2022-12-08:
Did you remember to do this:
```
tfodParameters.isModelTensorFlow2 = false;
```

Tarkkett, 2022-12-08:
That model was made with tensorflow2, if you set `tfodParameters.isModelTensorFlow2 = false;` it will load, but won't detect anything, i don't know if the model even load propelly. You need to set it to true;

> Did you remember to do this?
> 
> ```
> tfodParameters.isModelTensorFlow2 = false;
> ```

Tarkkett, 2022-12-08:
I am also giving you a wetransfer link, because it wouldn't open for me [here](https://we.tl/t-nLIXeY08Jy)

lizlooney, 2022-12-08:
I see two different types of crashes in the logs.
1. It was using tensorflow 2 code to load the model. If the model is tensorflow 1, you need to do `tfodParameters.isModelTensorFlow2 = false;`
2. The SoundPlayer was throwing an exception.
3.

Tarkkett, 2022-12-08:
@texasdiaz Looking at the crashlog, i see that the VuforiaFrameGenerator is failing and need Nomalization metadata. I generated the metadata, but it still crashes. In the new SDK documentation i saw that something was changed for the FrameGenerator. Do you have a clue what could be wrong?

lizlooney, 2022-12-09:
I have a clue what could be wrong!

It's because you have tfodParameters.isModelTensorFlow2 set to true!
Your log shows stack traces with `TfodFrameManager2.java`, which is only used for TensorFlow2.

Tarkkett, 2022-12-09:
> I have a clue what could be wrong!
> 
> It's because you have tfodParameters.isModelTensorFlow2 set to true! Your log shows stack traces with `TfodFrameManager2.java`, which is only used for TensorFlow2.

Im telling you again, that the model was made with tensorflow 2. It should be like that.

lizlooney, 2022-12-09:
Oh, sorry. My mistake.

Tarkkett, 2022-12-11:
@wangxdflight @lizlooney @texasdiaz @NoahAndrews Phefffff, after many failed attempts, i finally solved this problem. I had to add the metadate to the model, but also change the `org.tensorflow:tensorflow-lite-task-vision:0.2.0` to `org.tensorflow:tensorflow-lite-task-vision:0.4.3`. Please look into this. That line of code caused so much pain! Can the robot-controller ropository developers change this?

wangxdflight, 2022-12-11:
Glad it worked for you know. Anything forced you go use 0.4.3 version? (it is almost the latest.) What TF version did you use to train your model? For me, I used TF 2.9.2, and vision task 0.3.1 worked with it.

But I am having another issue, maybe people here can help me. It may related with compatibility caused by different versions too.
I am trying to do int8 quantiziation, and was able to generate tflite model for it, but having weird crashes on android (I am experimenting with a spearte android application, where I can easily change libraries to use.)
I tried both vision task API and interpreter API. float16 quantization works, but int8 doesn't.

kylejramstad, 2022-12-23:
> org.tensorflow:tensorflow-lite-task-vision:0.2.0

This worked. I have been making models with TFLITE Model Maker (which easy really really easy with Google Colab) and the robot controller app has been crashing all over the place.

It was crashing because the robot controller app is using an old version of org.tensorflow:tensorflow-lite-task-vision. In this old version it is looking for the output tensors in a very specific order. With TensorFlow 2.6, Google changed the order of those output tensors in the models.

If you build your own Robot Controller App (FtcRobotController), then you need to go to the build.dependencies.gradle file.
Change the line `implementation 'org.tensorflow:tensorflow-lite-task-vision:0.2.0'` into `implementation 'org.tensorflow:tensorflow-lite-task-vision:0.4.3'`, build the app, load it onto the control hub and everything works perfectly.

Windwoes, 2024-09-14:
TFOD has been removed from the SDK

closed 2024-09-14 by Windwoes