FIRST-Tech-Challenge/FtcRobotController #507: Issues linked are swapped in 8.1.1 README release info

NoahBres, 2022-12-03:
[image]

The first two issues have swapped issue references

CalKestis, 2022-12-03:
lol, I think you're just going to have to live with this one. :-)

For those wondering why we don't just edit and commit, there are two primary reasons.

 - The goal in recent years has been to keep the commit history as clean as possible.  We want the last commit in the history to reflect the last version that was merged.    We feel that this aids team understanding of what is what in this repository.
 - The origin of all the files found here, including the README, is in the SDK development repository and any change here needs to be reflected there.  That is highly subject to human error and so we just don't do this.

And therefore the rule is that this repository is frozen in between releases.

NoahBres, 2022-12-03:
Oh yea not bothered by this at all 😄
Just wanted to give you a heads up + make a small item for your internal issue tracker to make a quick edit for later :)

BD103, 2022-12-05:
Darn, I should have seen this issue before I made a PR #510 😅

Anyways, the reasoning makes sense. Glad you guys are aware of it