FIRST-Tech-Challenge/FtcRobotController #539: Sending to Control Hub resets network info

alan412, 2023-01-01:
When I rebuild 8.1.1 with no changes on Android Studio and I send it to our Control Hub, it causes the SSID and password to be reset.  

To test this, I:
 - Factory reset the Control Hub (applying power while pushing the reset switch)
 - Change Control Hub SSID and password to be legal for FTC
 - Clone the FtcRobotController repo directly
 - Build and use adb to send wirelessly to control Hub.
 - In the middle of installing it, the SSID and password change causing the install to fail.  (If I leave the SSID and password as default then it works fine.)

cmacfarl, 2023-01-01:
Is your Control Hub OS up to date?   You can check by connecting it via USB to a laptop with the REV Hardware Client installed.

alan412, 2023-01-01:
Yes.  I did an install of 1.8.3

cmacfarl, 2023-01-01:
The firmware that runs on the Expansion Hub is a distinct and separate entity than the OS that runs on the Control Hub.

The latest ExH firmware is version 1.8.3, but it is not at all involved in the SSID and password reset.  That is the purview of the android operating system running on the Control Hub.  The latest version of which is 1.1.3.

alan412, 2023-01-01:
Yes.  I responded to this not at the computer.  The Control Hub is at 1.1.3.

—Alan

alan412, 2023-01-01:
Here is a screenshot of the various versions.
[image]

gearsincorg, 2023-01-01:
There are some situations where loading a new RC App on the Robot Controller can cause/require the current app to be uninstalled and the new app to be installed.  This is different from just loading a newly modified Android Studio app on the RC.

This may be required when downgrading the app or when some other conditions are met.  There is usually a popup that occurs on Android Studio informing you that it can't just over-write the current app, and it requires an extra "OK" step.  Do you get such a popup?

When this uncommon uninstall/install happens, some other settings are also deleted... eg: the "selection" of the current robot configuration, and the current SSID.  

It seems that this uninstall/install action is being triggered by the state of the RC after the reset to "factory" default.
It would not normally be such an issue if you were hard wired to the RC, since the change in SSID would not cause the disconnection.  

Question:  After the factory reset, is there any sign of an RC app on the Robot Controller?

AlecHub, 2023-01-01:
SSID and Password reset happened to us a while ago.  We just left it at factory default.  I will change to non-factory values and monitor to see what particular action is triggering the reset.

Android Studio **2021.3.1 Patch 1**
SDK **v8.0** 
Control Hub OS **v1.1.3**
Expansion Hub Firmware **v1.8.2**

@cmacfarl has there been a v1.8.3 firmware release?

texasdiaz, 2023-01-01:
> @cmacfarl has there been a v1.8.3 firmware release?

You can find the changelog for the hub firmware here:
https://docs.revrobotics.com/duo-control/managing-the-control-system/updating-firmware/firmware-changelog

You can find all of REV's most recent software releases here; when a release is cut, this is the most likely location it would be released to (if you want to watch this repo):
https://github.com/REVrobotics/REV-Software-Binaries

But to answer your question fully, there are no firmware releases beyond 1.8.2.

-Danny

alan412, 2023-01-01:
We do not get a popup on Android Studio.   The current robot configuration
is deleted as well.  (with our normal application.).  I didn't test to see
if after the factory reset if there was any sign of an RC app.   I can test
that.

--Alan

alan412, 2023-01-01:
Curiouser and curiouser....

I did a factory reset which did not remove the FtcRobotController application.   Just for kicks, I did an
```adb uninstall com.qualcomm.ftcrobotcontroller```
I then installed the generic FtcRobotController.   The weird part is that when it rebooted, it came back to 16072-B-RC instead of the default SSID and pw.

Now updating the application doesn't change the SSID/pw.   So it might have been something corrupted on the Control Hub.    ¯\_(ツ)_/¯