FIRST-Tech-Challenge/FtcRobotController #617: New IMU API angular velocity wraps around

rbrott, 2023-03-02:
I have an op mode that essentially calls `mu.getRobotAngularVelocity(AngleUnit.RADIANS)` in a loop as voltage applied to the motors slowly ramps. Here's the plot:
[image]
I wouldn't expect the **velocity** here to wrap around.

I haven't had the chance to investigate in detail. I'm not sure if the issue is with the imu -> robot transform or is with the hardware (I'm actually not sure if this data is from a BNO or BHI).

gearsincorg, 2023-09-22:
Yes.  Weird that it seems to be wrapping at PI.  Might be normal for angle, but not angular velocity.

Windwoes, 2024-09-14:
This was fixed in v10.0

closed 2024-09-14 by Windwoes