FIRST-Tech-Challenge/FtcRobotController #646: Add an API for uploading robot configuration files

CyanBlob, 2023-05-09:
I'm working on a tool to simplify the programming process for teams without good development skills. 

As part of this tool I'm using the API to upload and build robot code. I would love to also have an API to upload robot configuration files so that students don't have to worry about keeping them up to date by themselves.

texasdiaz, 2023-05-09:
Hey Andrew! Can I ask what "API to upload and build robot code" are you referring to? I'm not presently aware of which API this might be.

CyanBlob, 2023-05-09:
Hey Danny. I'm using the API as documented in the OpenFTC project here: https://github.com/OpenFTC/Extracted-RC/blob/3f86727ccaa5424401ec941922e78f0bf50100cc/OnBotJava/src/main/java/org/firstinspires/ftc/onbotjava/OnBotJavaProgrammingMode.java#LL83C27-L83C27 

I know that there's not official documentation for the API, but I think that it would be beneficial.