FIRST-Tech-Challenge/FtcRobotController #649: Add DualShock sixaxis support

allenftc, 2023-06-07:
The DualShock 4 controllers have a six axis imu inside. This could be useful for teams just as another control method. There is more info here https://www.psdevwiki.com/ps4/DS4-USB.

Windwoes, 2023-06-07:
I believe that a rather significant issue here is the lack of sensor fusion on the controller for integrating the rate data.
Simply sending raw rotation rate data is probably not very useful.

j5155, 2023-12-06:
I believe this issue should be reopened as the issue is not solved.
It looks like sensor fusion to integrate the rate data is done in Cemu through the Mahony system
https://github.com/cemu-project/Cemu/blob/main/src/input/motion/Mahony.h#L17
https://hal.science/hal-00488376/document