FIRST-Tech-Challenge/FtcRobotController #696: AprilTag Quaternion Typos

FromenActual, 2023-09-14:
In `AprilTagGameDatabase.java`, the following tags have typos in the quaternion definitions:

```
                .addTag(7, "RedAudienceWallLarge",
                        5, new VectorF(-70.25f,-40.625f,5.5f), DistanceUnit.INCH,
                        new Quaternion(0.7071f,0,0,-7.071f,0))
                .addTag(8, "RedAudienceWallSmall",
                        2, new VectorF(-70.25f,-35.125f,4f), DistanceUnit.INCH,
                        new Quaternion(0.7071f,0,0,-7.071f,0))
                .addTag(9, "BlueAudienceWallSmall",
                        2, new VectorF(-70.25f,35.125f,4f), DistanceUnit.INCH,
                        new Quaternion(0.7071f,0,0,-7.071f,0))
                .addTag(10, "BlueAudienceWallLarge",
                        5, new VectorF(-70.25f,40.625f,5.5f), DistanceUnit.INCH,
                        new Quaternion(0.7071f,0,0,-7.071f,0))
```

The last terms should be `0.7071` not `7.071`

texasdiaz, 2023-09-14:
Thank you, we'll double check this and fix this in the next update.

WestsideRobotics, 2023-09-14:
@FromenActual 

As a workaround, teams can use a custom AprilTag library with your own Metadata.

FromenActual, 2023-09-15:
> use a custom AprilTag library

Yep! I will use that in the meantime.

@texasdiaz 

On a related note, can I ask about the rotations used for the AprilTag definitions? They're not what I expected.

The [official AprilTag coordinate system definition](https://github.com/AprilRobotics/apriltag#coordinate-system) states that when looking at a tag, the x-axis points to the right, y down, and z into the tag. The [official FTC field reference frame definition](https://ftc-docs.firstinspires.org/en/latest/game_specific_resources/field_coordinate_system/field-coordinate-system.html) states that from a red driver's perspective, the x-axis points to the right, y is forwards, and z is upwards. This implies that with no rotation (quaternion of `(1, 0, 0, 0)`), the tag would be facing downward, with the top of the tag closer to the red alliance station.

Tags 7-10 use a quaternion of `(.707, 0, 0, -.707)`, which corresponds to a -90 degree rotation about the z-axis. That means the tags are facing downwards, with the top of the tags closer to the audience, which is not how the tags are actually oriented in reality. Similarly, tags 1-7 use a quaternion of `(0.683, -0.183, 0.183, 0.683)`, which corresponds to a +90 rotation about the z axis followed by a -30 degree rotation about x (extrinsic). That means the tags are facing downwards, top away from the backdrops, and rotated 30 degrees towards the red driver station; again, that's not correct.

These are the quaternions I'm using:

* Tags 1-6: `(0.3536, -0.6124, 0.6124, -0.3536)`
* Tags 7-10: `(0.5, -0.5, -0.5, 0.5)`

I've confirmed these work when going through a homogeneous transformation from the `rawPose` to compute the robot pose on the field. Can the CENTERSTAGE tag orientations be redefined to use these values?

Thank you!

texasdiaz, 2023-09-20:
That's an interesting question about the starting orientation of the AprilTag. When we defined the starting orientation for the Vuforia targets, we started off assuming that the images were face-down on the mat to match an orientation of the phone cameras themselves starting face-down on the mat (so you can see the screen). However, in a world where there are no phone cameras (as the world will be next year, which is how we chose to define the field for AprilTags) the orientation of the tag was chosen to match the orientation of a webcam, which would not be facing down towards the floor but instead be upright facing in the positive Y direction (so the tag printing itself is facing away from the red wall, standing up in an upright orientation). That was the impetus of how the tag orientations were decided. 

Also please note that in [this document](https://ftc-docs.firstinspires.org/en/latest/apriltag/understanding_apriltag_detection_values/understanding-apriltag-detection-values.html) we define **ftcPose** which is not equal to the AprilTag coordinate system definition. Note in that document that:

```
Fun Fact: If the camera is pointing forward, the [ftcPose] X, Y & Z axes are consistent with the Robot Coordinate system.
```

Given this, does this make sense?

FromenActual, 2023-09-22:
Thanks for the response, I appreciate you taking the time to discuss this! I apologize in advance for my essay below 😉

Yes, your explanation makes sense to me, and I understand it. Assuming the tags start in that orientation (upright, facing away from the red drivers), then Tags 7-10 have a correctly defined orientation. However the orientation for Tags 1-6 are still wrong, and I disagree with defining the initial tag orientations like that.

## Tags 1-6 have incorrect orientations

Below is a crude demonstration I made in Blender (I've barely used Blender, so please bear with me 😃). The tag starts out in the orientation you describe (upright, facing away from the red drivers), then I applied the quaternion defined for Tags 1-6 `(0.683, -0.183, 0.183, 0.683)`. This is the resulting orientation when placed up against the backdrop:

[image]

[image]

Obviously, this is wrong. If the quaternion is changed to `(0.683, 0.183, 0.183, 0.683)` (just make the x-component positive), then it's oriented correctly (again, assuming the tags start upright facing away from the red drivers)

[image]

[image]

## Starting orientation of tags

I realize that this may become a discussion of preference or opinion, and we can disagree about what is "best". However I would strongly push for the initial orientation of the tags to be defined as face down, with the top towards the red drivers. My arguments are below (numbered, in case we need to reference them):

1. It's better to use existing standards than to create new ones ([XKCD](https://xkcd.com/927/) is topical here).

    * The [official axis definitions for AprilTags](https://github.com/AprilRobotics/apriltag#coordinate-system) already exist and are widely used. With no rotation (ie. the initial orientation), the tag axes _must_ match the [field axes](https://ftc-docs.firstinspires.org/en/latest/game_specific_resources/field_coordinate_system/field-coordinate-system.html); that means the tag is face down, with the top towards the red drivers.

2. The `rawPose` data uses standard axis definitions for the camera and tags.

     * `rawPose` is what comes directly out of the [official AprilTag pose estimator](https://github.com/AprilRobotics/apriltag/blob/b7bd75ca45d62ac431f10b78b69ea86198cbad14/apriltag_pose.c#L528), which uses standard axis definitions for the tag and camera.

3. The `rawPose` is the best way to compute the robot's location on the field.

    * Using homogeneous transformations makes this code really straightforward (as demonstrated by the [Vuforia implementation](https://github.com/OpenFTC/Extracted-RC/blob/ca2f581525ca463dad6d3cce1fa7a8f62859463a/RobotCore/src/main/java/org/firstinspires/ftc/robotcore/external/navigation/VuforiaTrackableDefaultListener.java#L394-L400), basically just need to replace `vuforiaCameraFromTarget` with `rawPose`). If non-standard axis definitions and starting orientations are used, then the computed robot location and orientation is completely wrong (exactly what happened to me when I tried it).

## Other thoughts

4. I think it's important to note that the tag's `fieldLocation` and `fieldOrientation` will likely only be used by advanced teams, who are experienced enough to wrap their heads around less "intuitive" but more standard axis definitions (eg. tags start facing down, and the camera y-axis actually points downwards).

    * I expect most teams will navigate towards AprilTags like demonstrated in the [example OpModes](https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/external/samples/RobotAutoDriveToAprilTagOmni.java). In this use case, the tag's `fieldLocation` and `fieldOrientation` are completely irrelevant. However in more advanced use cases (eg. computing the robot location and orientation), it's critical that these values  are well defined; IMO, it will be much better to define them based on the official coordinate systems that already exist.

5. There are multiple different coordinate systems we're discussing here, and we should be careful to not mix them up.

    * In your message, you said "we define ftcPose which is not equal to the AprilTag coordinate system definition". The way I read it, that seems to imply that `ftcPose` is a redefinition of the _AprilTag's_ axes, but it's not; it's a redefinition of the _camera's_ axes.

6. Also good to note (at least for myself) that `rawPose` and `ftcPose` are unrelated to the tag's `fieldLocation` and `fieldOrientation`

    * `rawPose` and `ftcPose` describe the location and orientation of the tag relative to the _camera's_ reference frame, whereas `fieldLocation` and `fieldOrientation` describe the location and orientation of the tag relative to the _field's_ reference frame.

7. The starting orientation of the cameras and tags have nothing to do with each other.

    * You mentioned the Vuforia targets started "face-down on the mat to match an orientation of the phone cameras themselves starting face-down on the mat"; this correlation between the camera and Vuforia target starting orientation is purely coincidental, because the standard axis definitions for both just happened to make them align with each other.

    * Also, I believe the Vuforia targets actually started face up. Below is how Vuforia defined the axes for their image targets ([source](https://library-archive.vuforia.com/articles/Solution/Working-with-Vuforia-and-OpenGL-ES.html), this was oddly difficul [... cut ...] s at a disadvantage. IMO the floor just got lowered, which is opposite to what FIRST has been working towards over the years.

9. "in a world where there are no phone cameras (as the world will be next year"

    * As an FTA, I actually appreciate this decision! I'm sure some teams won't be happy about it, but I think consolidating everyone to a more robust platform is the right move for the long run. I look forward to no longer fussing with flakey OTG cables in heavily abused Micro B USB ports!

texasdiaz, 2023-09-22:
See here, kids? This is what's called "Making your case" - and this is me saying, "THANK YOU" for providing such a compelling argument in such an easy to follow way (and laying down some "fat facts" as my daughter says). We will update the AprilTag Quaternions to your originally suggested values with our gratitude. We plan on releasing an update in the next couple weeks that will contain this change.

- Tags 1-6: (0.3536, -0.6124, 0.6124, -0.3536)
- Tags 7-10: (0.5, -0.5, -0.5, 0.5)

> Is there any chance the AprilTag processor will be updated to compute homogeneous transformations to determine the robot's location and orientation like Vuforia used to?

This is something we considered heavily for this season, but given that we added AprilTags so late in the CENTERSTAGE design phase, we couldn't place the AprilTags in a place where occlusions couldn't be avoided, and especially being the first season with AprilTags, it quickly sank in our priority queue. This is something we absolutely want to add in a future season, especially as we brainstorm better ways to integrate technology into the field design earlier in the game design process.

-Danny

FromenActual, 2023-09-22:
> We will update the AprilTag Quaternions to your originally suggested values with our gratitude. We plan on releasing an update in the next couple weeks that will contain this change.

Excellent, this is great to hear! Thank you so much! 

> This is something we considered heavily for this season, but given that we added AprilTags so late in the CENTERSTAGE design phase, we couldn't place the AprilTags in a place where occlusions could be avoided, and especially being the first season with AprilTags, it quickly sank in our priority queue. This is something we absolutely want to add in a future season, especially as we brainstorm better ways to integrate technology into the field design earlier in the game design process.

Understood, thanks for the clarification and reasoning behind the decision! That all makes sense to me, I look forward to seeing that added in the future!

-Dryw