FIRST-Tech-Challenge/FtcRobotController #700: Issue installing both FTC Driver Station and FTC Robot Controller Android apk(s)

GIMSRobotics, 2023-09-15:
Hi, 

We are a new team where our equipment came from a bin in 2019 and are forced to use the android app on an approved device:

I have six phones that are XT1609 Motorola G4 play -- I am unable to sideload the files either from the REV windows app or by sideloading in android. When it's sideloaded from android then it gives a parser error when trying to install through file manager. I have checked the install from unknown sources. When I try to transfer the apks from the REV app -- I get an error that says:

pkg: /data/local/tmp/9.0-robot-controller-app-RevHub-51.apk Failure [INSTALL_FAILED_OLDER_SDK]

The phones are running Android 6.0.1 -- I have searched and cannot find a solution that's realistic to update to Android 7.1.1 if it even is supported. I can't find more information about G4 phones being supported as far as the android versions it needs. I don't think Motorola even supports these anymore. Is there anyone that can help? At this point we don't have the money for a driver hub and control hub and haven't heard back from a grant. Thank you!

WestsideRobotics, 2023-09-15:
Sorry to say, the Moto G4 Play situation does not have a good solution.  Some of those phones can update to 7.1.1 and some cannot (despite many efforts).

You have probably reviewed these already, but the relevant sections of Game Manual 1 are: RE07 (inlcuding footnotes) and RS03.  The FTC SDK does not support lower Android versions.

You are very likely to get a positive response from other FTC teams in your region.  Some will have Android phones to lend or donate to your program, to get through the season.

GIMSRobotics, 2023-09-15:
OK I wanted to be sure I wasn't missing anything. Thank you for the response. I did try everything as far as Moto drivers, and RSD lite etc type methods. I wasn't successful. 

I will use this information to pursue some donor phones in my region and follow up on the grant request. I am sure we can find a solution to get us though.

closed 2023-09-15 by GIMSRobotics

NoahAndrews, 2023-09-15:
I can't endorse the "stock ROM" site that this links to, but someone did manage to get theirs updated: https://www.chiefdelphi.com/t/flash-android-7-on-motorola-moto-g4-play-phones-that-wont-upgrade/440746

GIMSRobotics, 2023-09-15:
I think I may have been able to do that, but it seems like you have to get an unlock code from Motorola first to unlock the bootloader. I did fastboot oem get_unlock_data and entered that code into Motorola's bootloader site and they say that this device is not eligible. I am going to look toward other solutions. Thanks for the help!

NoahAndrews, 2023-09-15:
Ah, if you can't unlock the bootloader you're almost certainly out of luck with the device entirely.

GIMSRobotics, 2023-09-15:
***UPDATE*** 

I was able to find a solution by downloading this 'Motorola_Moto_G4_Play_XT1609_HARPIA_VZW_NPIS26.48-43-2_cid2_subsidy-VZW' and then use 'RSD_Lite_Motorola_XML_To_Batch_Script' to create some .bat files --- you have to run the .bat files NOT as administrator it wasn't working with. I had an issue with a Verizon bootloop at the loading screen, but after a few minutes and rebooting twice (happened on both) it resolved itself. I did try to wipe data and wipe cache again. I have successfully completed moving from 6.0.1 to 7.1.1 on 2 of my 6 devices now. BOTH of the FTC apps appear to work. I had no need to do anything with the bootloader. Success!! Thanks! @NoahAndrews! @WestsideRobotics

WestsideRobotics, 2023-09-15:
@GIMSRobotics 

That's some **hero** work right there.  Well done.

You are encouraged to post the full details (skipping no steps) here or at [**ftc-community**](https://ftc-community.firstinspires.org).

Or, email that to [**me**](https://github.com/WestsideRobotics) and it can be shared more broadly (with credit of course!) as a potential technical resource.

It's true that Android phones will be phased out for competition next year, but that announcement applied only to RC phones, not DS phones.  The Moto G4 Play had the largest installed base, as the default phone model offered in the FIRST Storefront for multiple years.

GIMSRobotics, 2023-09-15:
@WestsideRobotics Thank you so much!!! I will draft a more detailed email and send that to directly at the address listed on your main page. Maybe if this solution gets out there it can help other teams that maybe thought they were stuck due to the same issue.

texasdiaz, 2023-09-16:
I'm curious, did you try using the tool listed in the Game Manual 1? Did that not work?

GIMSRobotics, 2023-09-16:
I didn't try the rescue tool until after I already had solved my issues. I did install the rescue tool after the fact and it looks like it could have worked to rescue it from 6.0.1, but they are on 7.1.1 and I don't want to try the tool just for fun. It's likely possible to bring them up to 7.1.1 in multiple ways though. @texasdiaz