FIRST-Tech-Challenge/FtcRobotController #710: symbol BuildConfig cannot be resolved in class FtcRobotControllerActivity

Vector5233, 2023-09-20:
Hi, we're using the 2023 season version of the SDK via the RoadRunner distribution.  The project builds successfully.  However, as we looked "under the hood" in the FtcRobotController/java/internal/FtcRobotControllerActivity.java class, an error message appeared: symbol BuildConfig cannot be resolved in lines 82 and 406

82 is the import statement

import com.qualcomm.ftcrobotcontroller.BuildConfig;

406 is the usage.

We did not make any changes to the code; the problem happens on two separate "clean" installations.  The error message does not appear to interfere with the project build.  Just FYI.

NoahAndrews, 2023-09-20:
That typically means that you need to tell Android Studio to do a Gradle Sync.

Vector5233, 2023-09-21:
RESOLVED.

Thanks.  That was a clue enough to get resolution, as follows:

* Upgraded Android Gradle Plugin to 8.1.1.  Synced.  No change.
* Executed Build > Make Project.  Error disappeared.

Issue occurred and was resolved for both the RoadRunner distribution and a clean copy of the FIRST-Tech-Challenge distribution.