FIRST-Tech-Challenge/FtcRobotController #720: AprilTag Axes Rendered Backwards

FromenActual, 2023-09-23:
The [official axis definitions for AprilTags](https://github.com/AprilRobotics/apriltag#coordinate-system) is defined as +x to the right, +y downwards, and +z into the tag. It looks like this:

[image]

However, `AprilTagCanvasAnnotator.java` flips each axis when `drawAxes` is enabled. Not only is this wrong, it also implies a left-handed coordinate system (big no-no!):

[image]

The render is similarly flipped when `drawOutline` is enabled, the red and green lines should be joined in the top left corner of the tag, not the bottom right.

Since the orientations of the tags will be updated to be based on the standard AprilTag axis definition (as discussed in #696), it would help to have the axes rendered as such. I'll reiterate from [my other comment](https://github.com/FIRST-Tech-Challenge/FtcRobotController/issues/696#issuecomment-1730926274) that this does not conflict with the definition of `ftcPose`, because that's a redefinition of the _camera's_ axes, not the _AprilTag's_ axes. I also understand that having the z-axis rendered into the tag may look a bit confusing because of a lack of occlusion, but I think correctness is more important in this case. Plus, the axis rendering is disabled by default, so only people who need it will enable it.

Windwoes, 2023-09-23:
At one point we tried to make the axes "more correct" but it resulted in needing an axis pointing into the image instead of out of the image and that looked very visually "wrong".... also, this was the background behind why it's disabled by default:

[image]

FromenActual, 2023-09-23:
Thanks for the background context! I do actually agree with having the axis rendering disabled by default, since most people don't need it, and it can sometimes get in the way. The border outline is sufficient for most users just to know if a tag has been detected.

However, in the cases when users enable the axis rendering, they're probably doing so in order to understand the orientation of the tags. I know that for myself, I initially wasted time debugging because I didn't realize the rendering was flipped, and even after I knew that, I still found myself getting mixed up. So in order to prevent more people from getting confused, I really think the axes should be rendered correctly.

I agree about rendering the z-axis into the tag looking visually "wrong". I have a few suggestions on that:

1. Render the z-axis into the tag, and just accept that it looks a bit weird. Most people won't enable the axis rendering anyways, and the few people that do can probably wrap their heads around it.
2. Render the z-axis into the tag, and set the alpha value for the `blueAxisPaint` to make it partially transparent. That may help it look a bit less weird, it's sort of like a partial occlusion.
3. Just don't render the z-axis. Strictly speaking, you only need 2 axes to properly define a 3D coordinate system. This would ensure it doesn't look wrong while also not being incorrect.

I don't really have a strong preference for how the z-axis is rendered, as long as it's not incorrect.

Windwoes, 2024-09-14:
This was resolved in v10.0

closed 2024-09-14 by Windwoes