FIRST-Tech-Challenge/FtcRobotController #724: SDK 9.0 "Driver Station is Obsolete" warning message if date is greater than October 1, 2023

texasdiaz, 2023-09-25:
**What:** Beginning in October 1, 2023 Driver Stations will begin showing a "Warning: Driver Station Software is Obsolete" message.

**Why:** There are two reasons why a Driver Station might show a warning that it is obsolete even though it is at the most recent version of the software:

1. The time and date in the Driver Station is set into the future. Driver Hubs sold by REV have traditionally had a default date set by the manufacturer several years into the future, which triggers the obsolescence logic which looks at the build time of the DS and RC apps and compares it to the date and time set on the device. If these are too far apart, the Driver Station App can provide a false-positive warning that the app is obsolete.

2. SDK 9.0 has a build date of 8/30/2023, which was the earliest any version of the SDK has ever been built that did not flag a rebuild during additional testing. Unfortunately the logic that calculates obsolescence requires that, starting on October 1st of each year, the "non-obsolete" version of the software must have a build date of 9/01 or greater of the same year (rolling calendar year). This has historically never been a problem, until now. 

**Resolution:** In order to resolve this issue, two action items are required:

1. Set the Driver Station device's date and time correctly.
2. Update to SDK 9.0.1 or greater.

NOTE: SDK 9.0.1 will be releasing 9/29/2023.

texasdiaz, 2023-10-03:
Resolved #731

qwertychouskie, 2023-10-27:
This issue should probably be closed now.

closed 2023-10-29 by texasdiaz