FIRST-Tech-Challenge/FtcRobotController #744: Error getting native address from native library: task_vision_jnl

DominantDuck, 2023-10-04:
I am having trouble running my OpMode with a custom tflite file; I keep getting the message "Error getting native address from native library: task_vision_jnl" on my driver hub after pressing init.  My "model_unquant.tflite" is located FtcRobotController -> assets.

Here is some of my code: 
` private static final String TFOD_MODEL_ASSET = "model_unquant.tflite";`

```
 tfod = new TfodProcessor.Builder()

                .setModelAssetName(TFOD_MODEL_ASSET)

                .setModelLabels(LABELS)
                .setIsModelTensorFlow2(true)
                .setIsModelQuantized(true)
                .setModelInputSize(300)
                .setModelAspectRatio(16.0 / 9.0)

                .build();`
```

DominantDuck, 2023-10-04:
I used Google Teachable Machines.

texasdiaz, 2023-10-04:
Is there something wrong with the ftc-ml tool? I'm just curious what your hesitation is.

You can use Google CoLab to create object detection models if you want, but we do not provide support for it. The only tool we support directly is ftc-ml.

DominantDuck, 2023-10-05:
My only hesitation is the coach uploading the video files; it would be much easier for me to upload the files, especially after-practice hours. Thank you for your help.

closed 2023-10-05 by DominantDuck