FIRST-Tech-Challenge/FtcRobotController #745: Calibration parameters for HD Pro Webcam C920 not automatically detected

MafteiAlbert-Alexandru, 2023-10-04:
When using a HD Pro Webcam C920 with the Vision Portal, the calibration parameters are not automatically detected. This happens due to a "mishap" from Logitech who has the same product registered under **3** different product ids:
HD Pro Webcam C920 - 0x82D, C920 PRO HD Webcam - 0x08E5, C920 HD Pro Webcam  - 0x0892(Note: I have the one with 0x0892). 

For most teams the workaround is obvious: Copying the calibrations, changing the ids, and adding them to teamwebcamcalibrations.xml, but it would be ideal if the other product ids would also be registered in the SDK, in order to avoid confusing rookie teams(especially since it's stated that the SDK has the calibrations parameters for their camera).

texasdiaz, 2023-10-04:
Thanks for the feedback. We are aware of the multiple C920 product ID issue, but we're unable to get a straight answer from Logitech regarding the difference in the devices or changes in the optics. We have also been unable to get hands on the device to verify that the calibration in the SDK is valid for all product IDs. The only way to properly verify is to perform the camera calibration process and compare the results.

- https://ftc-docs.firstinspires.org/en/latest/apriltag/vision_portal/apriltag_camera_calibration/apriltag-camera-calibration.html

Are you able to perform this task and provide us with your results?

MafteiAlbert-Alexandru, 2023-10-04:
> Are you able to perform this task and provide us with your results?
I will to do the calibration and come back with the results.

texasdiaz, 2023-10-12:
What resolution did you calibrate the camera at? Each resolution will have different calibration parameters.

MafteiAlbert-Alexandru, 2023-10-12:
Oops, forgot to mention. It was 800x600

texasdiaz, 2023-10-20:
This has been added for consideration for the 9.1 release, currently targeted for January 2024. This issue will remain open until this work has been completed.

BradDGit, 2023-12-15:
I have a with the PID 0x08B6. I have run through calibrations and here is the data the I got back withb comparisons against the exist FTC values.

Resolution | Source | fx | fy | cx | cy
-- | -- | -- | -- | -- | --
640x480 | FTC | 622.001f | 622.001f | 319.803f | 241.251f
640x480 | My Camera | 638.991 | 638.991 | 327.391 | 246.924
  |   |   |   |   |  
800x600 | FTC | 775.79f | 775.79f | 400.898f | 300.79f
800x600 | My Camera | 791.489 | 791.489 | 400.403 | 313.946
  |   |   |   |   |  
640x360 | FTC | 463.566f | 463.566f | 316.402f | 176.412f
640x360 | My Camera | 477.753 | 477.753 | 320.357 | 181.369
  |   |   |   |   |  
1920x1080 | FTC | 1385.92f | 1385.92f | 951.982f | 534.084f
1920x1080 | My Camera | 1439.292 | 1439.292 | 947.996 | 550.08
  |   |   |   |   |  
800x448 | FTC | 578.272f | 578.272f | 402.145f | 221.506f
800x448 | My Camera | 601.618 | 601.618 | 392.459 | 225.876
  |   |   |   |   |  
864x480 | FTC | 626.909f | 626.909f | 426.007f | 236.834f
864x480 | My Camera | 644.059 | 644.059 | 425.96 | 241.591

I am not that familiar with running the process or interpreting the results so may process may not have been ideal but it looks like my values are somewhat similar to the ones in the FTC files so most likely this PID could just be added as a new PID for the existing values.

I also found an interesting article that discusses the different C920 models.
https://www.streamtechreviews.com/blog/c920-variations

Please let me know if there is additional data I should be trying to get.

BradDGit, 2024-01-05:
@texasdiaz Is

> This has been added for consideration for the 9.1 release, currently targeted for January 2024. This issue will remain open until this work has been completed.

Is this still targeted for this month. Will the PID I referenced above also be included?