FIRST-Tech-Challenge/FtcRobotController #749: Breaking changes for OAK-1 Lite introduced with CenterStage

Powercube7, 2023-10-06:
Hello there!

With the addition of the VisionPortal in FTC CenterStage, our team has noticed breaking changes in the robot controller's code while attempting to install the DepthAI library required for the OAK-1 Lite vision sensor.
**What:** The robot controller application crashes when booted after installing the DepthAI library.
**Why:** It appears that there is a conflict between OpenCV classes in the code of the robot controller and the code of the [bytedeco/javacpp-presets](https://github.com/bytedeco/javacpp-presets) repository. Building the project generates errors because of this issue. Attempting to exclude the library from the module causes crashes whenever the app is launched.

### Attachments:
- A [logcat](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/12833356/Logcat.txt) containing 3 instances of the RC crash
- The [build.gradle](https://github.com/FIRST-Tech-Challenge/FtcRobotController/files/12833369/build.gradle.txt) file containing DepthAI and JavaCPP (OpenCV module excluded; remove the `exclude` statement to observe duplicate classes errors)

### Additional information:
- Tested devices: OPPO A54, Control Hub
- No crash occurs on the 8.1 version of the RC
- RC version: 9.0

Windwoes, 2023-10-06:
Hi there - if there's an incompatibility in the versions of OpenCV used, probably your best bet will be to exclude the entirety of the Vision module from the Robot Controller build.

Powercube7, 2023-10-08:
@Windwoes Thank you for the commment! Is there any resource on how I would be able to do that? Would doing so break the FTC VisionPortal?

Windwoes, 2023-10-08:
It would essentially just be commenting out this line:

https://github.com/FIRST-Tech-Challenge/FtcRobotController/blob/master/build.dependencies.gradle#L15

And then deleting all the vision portal sample OpModes from the samples directory.

I do not think it would break the robot controller build but I'm not sure. The blocks module does depend on the vision module, but it's already been compiled into an artifact here so I think you could get away with it. Obviously trying to use vision portal from inside blocks would crash

As an aside, it may be worthwhile to confirm on the Q&A forum whether the OAK1 is legal for competition.

closed 2024-09-14 by Windwoes