FIRST-Tech-Challenge/FtcRobotController #763: LiveView no longer works if push new code to Control Hub while the OpMode is still runniing.

trc492, 2023-10-11:
If I ran a vision opmode but did not stop the opmode and then push new code to the Control Hub, then the new code will not show LiveView. The Control Hub display shows the Robot Controller app with white view as if there was no vision at all. I will not be able to get LiveView to work again until I power cycle the Control Hub. I also noticed when this happened, the webcam (Microsoft LiveCam HD 3000)'s blue light was not ON as if it failed to open the USB webcam. If I stop the opmode first before pushing new code, then this won't happen as if not stopping the opmode would skip some sort of cleanup so that the camera will be in a bad state causing a failure in opening the camera the next time. BTW, if it helps, I never experienced this issue until I started playing with switchable camera. Don't know if it is related. Also, to rule out that our code caused this. I did the same experiment but with the ConceptAprilTagOptimizeExposure sample OpMode. With the opmode still running, I recompiled and pushed the code to the Control Hub again. The Control Hub terminated the OpMode then hung (i.e. the new OpMode never run). I had to power cycle the Control Hub to get it back to normal.

trc492, 2023-10-11:
Further testing showed that this may be webcam specific. We got a new Logitech C920 and while this issue always happens to Life Cam HD 3000, it does not happen on C920 but it does occasionally hang if we push code without stopping the OpMode but not all the time.

trc492, 2023-11-02:
It looks like robotInit actually hung initializing the camera if we push new code to the Robot Controller without stopping the OpMode first. If we stop the OpMode then push new code, then everything works. It means stopping an OpMode must have executed some sort of cleanup code in closing the camera. If this is not done, the next time the camera is initialized, it hung.

hardcopi, 2024-07-17:
We had the same thing happen to us. We use the C270. Power cycling the robot fixed it though.