FIRST-Tech-Challenge/FtcRobotController #764: Get gain control returns a null pointer

trc492, 2023-10-11:
The following code caused a NullPointerException because gainControl was null, so I had to add code to test for null before using it.
    /**
     * This method returns the camera min and max gain setting.
     *
     * @return array containing min and max gain values, null if unsuccessful.
     */
    public int[] getGainSetting()
    {
        int[] gains = null;

        if (visionPortal.getCameraState() == VisionPortal.CameraState.STREAMING)
        {
            GainControl gainControl = visionPortal.getCameraControl(GainControl.class);

            if (gainControl != null)
            {
                gains = new int[2];
                gains[0] = gainControl.getMinGain();
                gains[1] = gainControl.getMaxGain();
            }
            // TODO: Figure out why gainControl is NULL?!
        }

        return gains;
    }

Windwoes, 2023-10-11:
It looks like switchable cameras are not currently set up to support controls besides focus and exposure. 

From `RefCountedSwitchableCameraImpl`

```java
    @Override protected void constructControls()
        {
        delegatingCameraControls.add(new SwitchableFocusControl(this));
        delegatingCameraControls.add(new SwitchableExposureControl(this));
        }
```

trc492, 2023-10-11:
How come the ConceptAprilTagOptimizeExposure sample OpMode did not get NullPointerException but our code did?

trc492, 2023-10-11:
Never mind. Because it wasn't doing switchable camera.

ftc19743, 2023-11-28:
We are seeing multiple related issues while utilizing 3 Logitech 270s:

- Attempts to set up each on their own VisionPortal hang indefinitely during the build process (drove us to switchableCamera)
- Inability to set gain on switchableCameras makes fine tuning exposure rates for AprilTags problematic (noted above)
- Setting exposure on the current camera sets the exposure on all the cameras (was this intended behavior?)
- Setting exposure ExposureControl.Mode back to Auto appears to have no effect on any of the cams.
 `    public void setAutoExposure() {
        // Wait for the camera to be STREAMING to use CameraControls (FTC SDK requirement)
        if (visionPortal.getCameraState() != VisionPortal.CameraState.STREAMING) {
            while (visionPortal.getCameraState() != VisionPortal.CameraState.STREAMING) {
                teamUtil.pause(20);
            }
        }
        ExposureControl exposureControl = visionPortal.getCameraControl(ExposureControl.class);
        if (exposureControl==null) {
            teamUtil.log("Failed to get ExposureControl object");
            return;
        }
        if (exposureControl.getMode() != ExposureControl.Mode.Auto) {
            exposureControl.setMode(ExposureControl.Mode.Auto);
            teamUtil.pause(50);
        }
    }`

WestsideRobotics, 2023-12-01:
> We are seeing multiple related issues while utilizing 3 Logitech 270s:
> Attempts to set up each on their own VisionPortal hang indefinitely during the build process (drove us to switchableCamera)

The [**MultiPortal**](https://ftc-docs.firstinspires.org/en/latest/apriltag/vision_portal/vision_multiportal/vision-multiportal.html) feature of the FTC VisionPortal **does handle three portals** normally.

To verify this, I started with the two-portal sample OpMode described [**here**](https://ftc-docs.firstinspires.org/en/latest/apriltag/vision_portal/vision_multiportal/vision-multiportal.html#test-opmode), and simply added a **third instance** of key variables and methods.

Here's the RC LiveView, courtesy of [**`scrcpy`**](https://github.com/Genymobile/scrcpy):

[image]

And here is the corresponding DS telemetry (merged three screenshots to list it all):

[image]

To help manage resources, this sample uses the compressed MJPEG format and a lower webcam resolution of 320x240.  The sample ran well, without needing to disable any streams (gamepad toggles in this OpMode).

The sample continued to run well, even after alllowing the default 640x480 resolutions.  The above images use 640x480, thus giving useable pose results (for the 2 calibrated webcams).  Naturally a competition scenario might require tighter controls including stream toggling.

Camera controls, including gain and exposure, should be available on each portal.

Note the calibration warning (red text), as one of the webcams was a Logitech QuickCam Pro 5000 for which the FTC SDK does not provide lens intrinsics.  The other two webcams were Logitech C270 and Logitech C920.  One was plugged into the Control Hub's USB 2.0 port, and the other 2 shared a powered USB hub in the 3.0 port.

The FTC Blocks code is posted [**here**](https://gist.github.com/WestsideRobotics/5e42fd30aa8ba737fa7e76309fbe5160), with image below (expand in new browser tab):

[image]

Java users can study the Blocks code, and easily follow along in OnBot Java or Android Studio.

============

All of this is not intended to diminish the valid points here about gain and exposure control under `switchableCamera`.  Just wanted to demonstrate that "MultiPortal" does fundamentally perform as expected.  Big points for robustness to developer @Windwoes.

Windwoes, 2024-09-14:
This was fixed in v9.2

closed 2024-09-14 by Windwoes