FIRST-Tech-Challenge/FtcRobotController #797: [build optimization] Don't build `armeabi-v7a` by default

qwertychouskie, 2023-10-27:
There are no FTC-legal `armeabi-v7a` (aka `arm32`) devices anymore, and EasyOpenCV supports `arm64-v8a` just fine nowadays.  Disabling the build of `armeabi-v7a` cuts the APK from ~60MB to ~50MB.

qwertychouskie, 2023-10-28:
Also `FtcCommon-9.0.1` includes a bunch of sound files that are not used.  Only these files are actually needed:

- ss_mf_fail.wav
- ss_light_speed.wav
- ss_mine.wav
- ss_bb8_down.wav
- ss_r2d2_up.wav

Everything else is not used, and just eating up space unnecessarily.  Removing these files should shave off another ~3.2MB from the source, and at least 1MB from the APK.

qwertychouskie, 2025-08-28:
I found another ~2MB savings:  The `.map` files for the various web stuff are shipped in the APK despite not being needed by teams (they are basically the web equivalent of debugging symbols, and teams aren't debugging the Javascript and CSS of the web pages hosted by the robot controller).

Excluding them is easy, just added this to the `android` section of `build.common.gradle`:
```gradle
android {
    ...existing stuff...
    aaptOptions {
        ignoreAssetsPattern "*.map"
    }
}
```

qwertychouskie, 2025-09-04:
BTW, I figured out what caused the significant regression in app deployment times in the Into The Deep season, it's an Android Studio regression in 2024.1.x.  Partial deploys worked in 2023.3.1.20, but do not in 2024.1.2.13.  I have a bug report open here: https://issuetracker.google.com/issues/442347342

Hopefully this gets fixed upstream in a timely manner, but any APK size reduction we can do helps mitigate the issue to a certain degree (and even when the upstream partial deploy issue is fixed, a smaller APK is still beneficial, though less marked of a difference).

Windwoes, 2025-12-19:
> Another more invasive optimization that can be done is to completely disable OnBot and Blocks. It cuts down the size a lot.
> 
> I did this by using [ExtractedRC](https://github.com/OpenFTC/Extracted-RC). I removed the dependency on the OnBotJava and Blocks projects from gradle, moved `assets/java` from RobotCore to OnBotJava, and removed the OnBotJava service from RobotCore's `AndroidManifest.xml`. I also had to patch `FtcRobotControllerActivity.java` to not reference OnBot or Blocks.

You've basically just described [OpenRC Turbo](https://github.com/OpenFTC/OpenRC-Turbo). After interest dropped off I stopped maintaining it, though.