FIRST-Tech-Challenge/FtcRobotController #841: OnBotJava font size setting seems to have no effect

pmichaud, 2023-11-20:
I don't know when this changed, but the "Font Size" setting in the OnBotJava editor no longer seems to have an effect.

I've tried changing the font size to values as low as 12 and as high as 40 and the font always appears to be the same size on my browser (Google Chrome / Kubuntu Linux).

I know the font size setting worked in previous versions of OnBotJava.  Being able to set a large font size is important whenever I'm trying to teach/present OnBotJava programming in classroom or online workshop environments (e.g. for projection).

Thanks,

Pm

dmssargent, 2023-11-20:
I can confirm that’s a real bug in the current system. We’ll mark that to be fixed in a future release.

What Robot Controller deployment model are you using? OnBotJava and Android Studio or OnBotJava downloaded from the official sources?

One more question, since, it’s also broken at the moment, is did you ever notice what happened with the font size when you adjust the window in previous versions of OnBotJava? Do you miss that behavior this year?

pmichaud, 2023-11-20:
Thanks for the confirmation -- I had been wondering if it was "just me" somehow.

The font-size setting seems to have no effect in either deployment model.  I have some Control Hubs that have OnBotJava installed via the REV hardware client, and other Control Hubs that have the app installed from Android Studio; both exhibit the same font size regardless of the font size specified in OnBotJava settings.

My recollection is that changing the size of the window didn't affect the font size at all -- in short, it did exactly what I expected (which is to keep the font-size as specified, and not to change it based on the size of the window).

Thanks!

Pm

dmssargent, 2023-12-01:
>My recollection is that changing the size of the window didn't affect the font size at all -- in short, it did exactly what I expected (which is to keep the font-size as specified, and not to change it based on the size of the window).

Thanks for the info! That's appreciated so we can fix this so it behaves as you expect.

Do you need a workaround in OnBotJava until we have a proper bug fix released?

pmichaud, 2024-01-04:
I meant to answer this earlier -- I don't immediately need a workaround.  I can cope well enough until a proper bug fix is available.

Pm

dmssargent, 2024-01-08:
Thanks for the heads up:

If you end up needing a fix sooner rather than later, you can save the following as a bookmark to run when you need to fix OnBotJava font sizes:

```
javascript:((f) => (f.location.href.includes('/java/editor.html')) && f.$('.editor-font').css('font-size', `${f.env.settings.get('fontSize')}px`))(window.frames[0])
```