FIRST-Tech-Challenge/FtcRobotController #846: RobotAutoDriveToAprilTagTank.java and RobotAutoDriveToAprilTagOmni.java in external.samples bug fix

H2OLooBots, 2023-11-28:
Hey there!

I was viewing the code for RobotAutoDriveToAprilTagOmni when I saw this line of code:
```
private static final int DESIRED_TAG_ID = 8;     // Choose the tag you want to approach or set to -1 for ANY tag.
```

This is all fine and good, but when I went to run the code, it would look for any AprilTag, not just the tag with the desired ID. I looked at the code and saw this is the usage of `DESIRED_TAG_ID`:

```
if ((detection.metadata != null)
                        && ((DESIRED_TAG_ID >= 0) || (detection.id == DESIRED_TAG_ID))) {
                    targetFound = true;
                    desiredTag = detection;
                    break;  // don't look any further.
                }
```

This would mean any valid tag ID (greater than 0) would trigger the code. I believe the logic was supposed to be like this:

```
if ((detection.metadata != null)
                        && ((DESIRED_TAG_ID <= 0) || (detection.id == DESIRED_TAG_ID))) {
                    targetFound = true;
                    desiredTag = detection;
                    break;  // don't look any further.
                }
```

This would cause -1 to look for any tag, like intended in the comment in the definition of the line. How could this be resolved?

texasdiaz, 2023-11-28:
I believe you are still referring to the 9.0 version of this sample. It has been corrected already in 9.0.1 (your link to the sample in your issue points to the corrected sample).

Is this the case?

-Danny

AlphaRant9, 2023-11-28:
For some reason when I opened GitHub on my laptop it didn't show the fixed code. That resolved the issue, thanks!

closed 2023-11-28 by H2OLooBots