FIRST-Tech-Challenge/FtcRobotController #942: Bug in VisionPortalImpl.close()

25wangj, 2024-02-17:
When VisionPortalImpl tries to remove cameraMonitorViewId from viewsInUse, it uses "ArrayList.remove(int)" instead of "ArrayList.remove(object)". This causes an IndexOutOfBoundsError. Line 663 of VisionPortalImpl should change from:

```viewsInUse.remove(cameraMonitorViewId);```

to

```viewsInUse.remove(new Integer(cameraMonitorViewId))```

If other people think this is correct I will submit a PR.

Windwoes, 2024-02-17:
Ooops. Yes, that is correct. Unfortunately, the deploy model of the SDK does not facilitate the acceptance of PRs against this repo.

A workaround for now would be to (a) stop streaming but not actually close the portal if you're doing it to save resources, or (b) simply allow the built-in clean up to run its course.

Windwoes, 2024-02-25:
NB: this was missed in testing because the usage of `close()` in the samples did not trigger a crash display, which in and of itself is a bug. That is, this bug was masked in the samples by the presence of another bug.

Windwoes, 2024-08-11:
This was fixed in v9.2

closed 2024-08-11 by Windwoes