FIRST-Tech-Challenge/FtcRobotController #948: Driver Station crashing on startup

Hydroflame522, 2024-02-23:
The FTC Driver Station APK is repeatedly crashing on startup

Hydroflame522, 2024-02-23:
I can provide logs or device details requested

WestsideRobotics, 2024-02-23:
> The FTC Driver Station APK is repeatedly crashing on startup

A team-specific issue is best posted here:
https://ftc-community.firstinspires.org/

And yes, useful feedback is more likely with a full report of the complete situation: control system, unique hardware and behavior, software versions, error messages, links to log files, links to OpModes, etc.

Hydroflame522, 2024-02-23:
After installing the FTC Driver Station app (using the latest REV hardware client to install the latest driver station), the app automatically crashes on startup (the tablet is android 11 32bit)

[image]

i dont know where else to get the error logs other than adb logcat
this is just trying to get the app to open on this tablet, itll open on every other android device i try, just not this specific android tablet. if more/other specific logs are required to diagnose the situation i would appreciate if someone could assist me.

Windwoes, 2024-02-23:
Based on that log, It looks like that version of Android requires an extra permission in order to be able to turn on WiFi (something the DS does on startup)

gearsincorg, 2024-02-23:
Perhaps check to see if that tablet has any special security provisions that can be manually enabled at the app level.  
For info... who makes that tablet, or is it just a generic...

texasdiaz, 2024-02-27:
Please realize that any help or assistance being provided here is being done purely unofficially. Officially we do not support running the Driver Station App on any hardware not specifically deemed legal in competition by the Game Manual. FIRST Tech Challenge does not support every version or manufacturer's flavor of the Android Operating System specifically because there is no standard set for manufacturers. Every manufacturer has the capability of modifying the Android Operating System (and do!) and FIRST Tech Challenge does not have the resources to debug and support them all.

cmacfarl, 2024-02-27:
The exception being thrown is because this flavor of Android does not have permission to write to https://developer.android.com/reference/android/Manifest.permission#WRITE_SECURE_SETTINGS, which as the documentation suggests is not something not for use by third party applications, and yet, WifiManager.setWifiEnabled(), deeper in the android call stack, is attempting to do so.  

This appears to be a further tightening of the screws by Google w.r.t. Android 11 versus previous versions of Android.  As @texasdiaz says, we only officially support the devices legal in the game manual.  Any other device is YMMV.

closed 2024-09-14 by Windwoes